netzero-labs/netzero #3: "operational_mode" : "backup" missing.

rob0101, 2024-02-19:
You might want to add "backup" as an "operational_mode" option.

Charges the powerwall at 3.7kW from the grid plus any spare solar on top for 5kW max.

zigam, 2024-02-19:
Thanks for the note.  It's interesting that this option is still available, I remember Tesla removed it from the app a couple years ago.  Setting the mode does seem to cause some weirdness in the app -- the Operational Mode options will disappear, so you can't use the app to reset the mode.  I'm hesitant to add it for that reason, it might cause confusion.

Have you found the mode to be useful?  How does the charge rate compare to self-consumption with 100% backup reserve, or grid charging?

rob0101, 2024-02-19:
I find it useful, as do some others in Australia on wholesale pricing. The 3.7kW charge rate is predictable and not too aggressive. 

I don't like to use the reserve level to force charging. Bit of a hammer to adjust a screw.

zigam, 2024-02-20:
I added `backup` as a valid option for `operational_mode`.  I haven't documented it yet.

zigam, 2024-04-29:
I'm not sure documenting this makes sense given the impact on the Tesla app.  But I'm ok keeping it as an documented option for now if it provides value.

zigam, 2025-07-16:
Reopening this issue. Tesla fixed the issue where Operational Mode options disappear when the system is in Backup mode -- now it shows Self-Powered and Time-Based Control, with neither selected. This means it's possible to reset the state using the app. We'll add this is an option in Netzero in-app automations, so Powerwall can be charged at a faster rate.

zigam, 2025-07-20:
Backup mode is available in version 2.1.4:
https://docs.netzero.energy/docs/tesla/BackupMode

closed 2025-07-29 by zigam

himynamesned, 2025-09-14:
I disagree that backup mode is the quickest method (as detailed in the app under automations). I also disagree with the backup documentation on the NZ site that TBC is quickest, but only controlled by Tesla. 

You can set backup reserve to anything above the battery's current SoC and it'll charge at ~5 kW per battery.. proving you don't exceed grid limit for the site, set in the Tesla app. You can even set to 100%, even if intended to get to 85% and get the same results - a work around for the 81-99% Tesla limit...

Useful for free electricity sessions via OE or price arbitration- maximising TBC output. 

I have shown screenshots of these, and manual control, in my other suggestion threads ref free electricity sessions and arbitration.

zigam, 2025-09-14:
> You can set backup reserve to anything above the battery's current SoC and it'll charge at ~5 kW per battery.. proving you don't exceed grid limit for the site, set in the Tesla app. You can even [...]

That works sometimes (if you are in TBC mode with Grid Charging enabled), but we have enough data from users to know that it's not reliable. In this mode it's still TBC deciding to charge, nudged by the backup reserve change; and because TBC is not predictable, it can't be guaranteed to charge at the faster rate. If any of the conditions are changed--Grid Charging disabled, Self-Powered mode enabled, or TBC decides it's not time to charge--you will get the lower 1.8kW/PW charging rate.

Note: we're not claiming backup mode is the fastest charging method, just that it's the fastest **guaranteed** method.