netzero-labs/netzero #8: Change charge/Discharge limits

gnarmstr, 2024-04-02:
Hi,

Love your App. Are you able to change the charge and discharge limits for the Powerwall 2. My power walls have been restricted to 3kw for both charging and discharging and I want to put it back to 5 kw, but Tesla will not do it.

zigam, 2024-04-02:
Hi,

Thanks for the feedback!  There are no APIs available to directly control these settings.  Based on what I've seen, Powerwall charge/discharge depends on the mode it's in, and any limits determined by your utility (and managed by Tesla).

I believe when charging from the grid, time-based control will result in higher charge rate than self-consumption mode (~5kw vs. 3kW, assuming no limits in place).  When discharging to the grid, the limitation most likely depends on your utility.

If you have the local gateway connection set up, you can inspect the configured limits: Diagnostics / More / Select API call / system_status.

Can you describe your configuration (mode/grid charging/export mode) and charge/discharge rates a bit more?

gnarmstr, 2024-04-03:
I was told that it’s set by Tesla based on what the Utility company says. I don’t understand why setting the charge/discharge limit to 3kw has anything to do with the grid. Every other state in Australia isn’t limited.

I never charge from the grid, just from solar.
Not sure what information you are requesting in your last paragraph.

Geoff

rob0101, 2024-04-03:
Your installer should have set the limit based on what your network distributor (e.g. SAPN in South Australia) told them to.
Talk to your installer, they have access to the hidden settings.
It might be a misconfiguration or could be a limitation for your state, region or even your specific street.

Everyone in Australia has an export limit. You might not be exporting, but might have your Powerwall setup as if it does.
It's typically 5kW per phase.   Some regions allow 10kW in some neighbourhoods.

Recently the distributors have started to offer flexible exports up to 10kW per phase in return for them being able to curtail your exports when there's too much generation and not enough consumption (very common in SA).
People who choose not to opt into the flexible exports mechanism get put on a lower fixed rate, in SA the rate is only 1.5kW.
see: https://www.sapowernetworks.com.au/public/download/?id=324009
Again, doesn't sound like what you're trying to do, but might be how it is configured.

Your installer should be able to shed some light on what's happening.

closed 2024-05-03 by zigam