netzero-labs/netzero #9: Tesla Gateway Firmware 24.4.0 ... Are diagnostics now gone?

eJonny, 2024-04-10:
On Sunday morning I discovered an issue with my Tesla Energy system. As of 4:15am neither the Tesla app nor Netzero for Tesla were reporting data nor allowing system setting changes. I called Tesla solar support. After an hour on the phone with tier 1 support, trying a reset of the gateway, then a power down of powerwalls and reset of the gateway, followed by an entire power down of my home, tier 1 said that the issue would have to be escalated to tier 2 support.

Later Sunday afternoon I used the Tesla One (formerly Tesla Pro) app to login to the gateway and noticed that the firmware on the gateway was showing as 24.4.0 (which seemed new to me) and that the configuration of my system had been wiped (no grid code for starters).

I called tier 1 support again, explained what I saw, and they confirmed that a firmware update began at 4:16am. They updated the ticket with this information.

On Monday morning, apparently tier 2 support reconfigured my system as I noticed data again in the Tesla (and Netzero for Tesla) app. Tier 2 did not properly configure utility rates nor did they properly enable the grid charging option. I called Tesla solar support back and tier 1 said they would enable grid charging and export everything settings for my system, but it would take a few hours to reflect in my app.

As of late Monday, I'm back up and running. It appears that this firmware disables local access to the energy system. Navigating to the IP address of the gateway using a web browser no longer allows an option to log in.  Also, I no longer seem to be able to access diagnostics from the Netzero for Tesla iPhone app. Is this because of the changes Tesla implemented to local access? 

I'm running Netzero for Tesla version 1.72 iOS app.

zigam, 2024-04-11:
My gateway is on `24.4.0 0fe780c9` and I don’t have any issues with access.

What do you mean by “it no longer allows an option to login”?  You have to click Continue, scroll to the bottom and click Upgrade Later at the bottom. 

 It’s also possible your gateway ip address changed during this process.

eJonny, 2024-04-11:
[image]
[image]

I think these HTML GUIs changed from 23.44.0 to 24.4.0. I think the "UPGRADE LATER" option used to be on the first page. 

At any rate, with the confidence that there should be an "UPGRADE LATER" option, I discovered that the password I had set on the gateway was wiped out with the firmware update and I had to do a password reset. Once that was done, Netzero for Tesla was able to connect for diagnostics. 

Thank you for the feedback.

closed 2024-04-11 by eJonny