netzero-labs/netzero #10: Dynamic rate plan: allow no export tariff

jonputsman, 2024-04-27:
Hi,
I’m trying the dynamic rate plan feature with Octopus Agile. It seems you have to select both an Import and Export tariff from the lists. 

We have a legacy solar export contract called a “feed-in tariff”, which is not administered by Octopus. We get paid for what we generate, not what we actually export. So with these conditions we only want to export solar when the Powerwall is full. 

So it would be great if we could select “none” for the Export Tariff - or a fixed price of 0 - rather than having to choose one of the Octopus export tariffs.

Thank you.

zigam, 2024-04-28:
Thanks for the suggestion, I added a "No export tariff" option.  It's available in the [web app](https://app.netzeroapp.io) now, and will be available in app stores in the next couple days.

Give it a try and let me know how it works.

jonputsman, 2024-05-02:
Thank you. I picked up the updated iOS app today and have enabled the “No export” tariff. 

I saw a few gremlins after the update, but killing and restarting the app fixed / avoided them. 

For reference:
1. I got this after the update when I went into Utility Rate plan menu
[image]

2. I then signed out and after signing back in, no Utility Rate Plan item was visible:
[image]

zigam, 2024-05-02:
Thanks for the report!  I'll fix these issues in the next release.

zigam, 2024-05-03:
> When I try to paste my Amber api into the app, the text entry box is covered by the keyboard. 

Thanks, this will also be fixed in the next release.

jonputsman, 2024-05-05:
Octopus Agile with No Export tariff is working well, thank you for the updates.

Please close this issue when you wish.

closed 2024-05-05 by zigam