netzero-labs/netzero #11: FR: Ability to inject dynamic tariff via POST

purcell-lab, 2024-04-28:
Great work. 

I see you are adding dynamic tariffs for Octopus which is nice, but of course doesn't work for everyone. 

Could I suggest an improvement to be able to inject our own pricing via your API.

For example here is my pricing for the next 24 hours in 30 minute intervals, which I could easily POST via your API, which I presume you NetZero could then process and translate in a form suitable for upload into my powerwall utility rate plan via the FleetAPI.

```
'load_cost_forecast': [0.49, 0.42, 0.38, 0.37, 0.34, 0.21, 0.19, 0.2, 0.2, 0.18, 0.17, 0.15, 0.14, 0.15, 0.14, 0.14, 0.15, 0.14, 0.14, 0.14, 0.15, 0.19, 0.19, 0.21, 0.38, 0.24, 0.14, 0.13, 0.13, 0.11, 0.06, 0.06, 0.06, 0.09, 0.09, 0.1, 0.11, 0.11, 0.12, 0.15, 0.16, 0.19, 0.2, 0.35, 0.55, 0.56, 0.64, 0.58, 0.57, 0.55, 0.41, 0.4, 0.38, 0.24, 0.22, 0.24, 0.24, 0.21, 0.2, 0.2, 0.2, 0.2, 0.18, 0.15, 0.17, 0.17, 0.17], 
'prod_price_forecast': [0.25, 0.19, 0.14, 0.13, 0.12, 0.12, 0.1, 0.11, 0.11, 0.09, 0.08, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.1, 0.1, 0.12, 0.27, 0.15, 0.05, 0.04, 0.04, 0.04, -0.01, -0.01, -0.01, 0.01, 0.02, 0.02, 0.04, 0.04, 0.04, 0.07, 0.08, 0.11, 0.12, 0.12, 0.3, 0.32, 0.39, 0.33, 0.32, 0.3, 0.18, 0.16, 0.15, 0.15, 0.12, 0.15, 0.14, 0.12, 0.11, 0.11, 0.1, 0.11, 0.09, 0.06, 0.08, 0.08, 0.08]
```

rob0101, 2024-04-28:
Mark, I implemented code that mapped 30m prices into 1 of 4 tariff rates to give to the powerwall and it wasn't a great outcome.

Only having 4 price bands where each has its own FIT proved too restrictive for 30m changes. 

The TBC algorithm really struggled to do predictably sensible actions for me. The old Tesla API also had some rate limits that kicked in at the most inappropriate times. 

What's needed is for Tesla to accommodate highly variable pricing with individual FITs.

zigam, 2024-04-29:
@purcell-lab based on our previous conversations on TMC, I assume that's for Amber Electric?  I plan on adding support for Amber soon, so if that's something you'd want to test send me a note to ziga@netzeroapp.io.

@rob0101 it's possible to configure tariffs with more than 4 price bands, Netzero is currently doing this for Octopus Energy, see [Dynamic Electricity Pricing](https://github.com/netzero-labs/netzero/blob/main/docs/tariffs/Dynamic-Electricity-Pricing.md).

rob0101, 2024-04-29:
Ah, that's interesting.  How frequently does Tesla allow you to update the tariff ?

purcell-lab, 2024-04-29:
> @rob0101 it's possible to configure tariffs with more than 4 price bands, Netzero is currently doing this for Octopus Energy, see [Dynamic Electricity Pricing](https://github.com/netzero-labs/netzero/blob/main/docs/tariffs/Dynamic-Electricity-Pricing.md).

Lightbulb.

teslapy.Battery.create_tariff() allows you to create more than four BatteryTariffPeriod() and more than four BatteryTariffPeriodCost(), in fact you can specify 48 periods which can be loaded into your powerwall using battery.set_tariff().

The FleetAPI also seems to support the same tariff endpoint.

[image]

zigam, 2024-04-29:
> Ah, that's interesting. How frequently does Tesla allow you to update the tariff ?

I'm not sure what the rate limits are, but for Octopus Energy at least, the tariffs only really need to be updated a few times a day: once when next day's rates are announced, and then a few minor updates throughout the day.

I imagine it would work with Amber in a similar way, but using forecasted rates.

rob0101, 2024-04-29:
I'll have a play. 4 bands was way too much of a restriction

rob0101, 2024-04-29:
> I imagine it would work with Amber in a similar way, but using forecasted rates.

The Amber forecast rates can change every 5 minutes. I played with doing it every 15 minutes max, but mostly every few hours on average.

rob0101, 2024-04-29:
I was getting rate limited over a long period, eg too many requests over a 24 hour period  

I don't know if it was 24 hours or some other timeframe.

It's possible the new api has different rate limits.

zigam, 2024-05-03:
Amber Electric integration is now available in the Netzero app.

closed 2024-05-03 by zigam