netzero-labs/netzero #16: Grid charging option suggests its disabled instead of enabled

nickz98, 2024-05-15:
Using my ios version of your app.
Within your app the 'grid charging' option 
has a button that suggests the feature is disabled when in fact it's enabled.
If I correlate the option to the entry that is seen from the Tesla app you can see how its misleading.
I guess you just need to change the word 'Disabled' to 'Enabled'?

[image]
[image]

zigam, 2024-05-15:
Thanks for the report, the label is indeed incorrect.  I fixed this and it will make its way into the next release.

zigam, 2024-05-15:
Independently from the bug, I also realize a changing label next to a switch is confusing, I'll change that to a toggle similar to Tesla's.

zigam, 2024-05-17:
The label is fixed in version 1.82, available in the app stores.  You might need to update manually, since the version is still rolling out.

closed 2024-05-17 by zigam