netzero-labs/netzero #20: Roy

FynnGolfer, 2024-07-03:
Have been trying your link to Powerwall but frequently get no change of status. Is there a minimum interval between change requests, like 5 minutes or something?
Thanks

zigam, 2024-07-03:
Thanks for the report, can you provide some more details? Which link to Powerwall are you using and what do you mean by "no change of status"?

FynnGolfer, 2024-07-03:
I’m am using an API via Netzero to change Powerwall settings like ‘Backup Reserve’ and ‘Operational Mode’. Sometimes the request is actioned immediately on others it can take several minutes or I get fed up with waiting. I was wondering if there is a limit too the rate of requests either through Netzero or by Powerwall.Sent from my iPad

zigam, 2024-07-03:
There are two possible issues here:
1. Configuration changes can sometimes take a few minutes to take effect -- they have to be sent to the Tesla Gateway. You can confirm this by manually changing a configuration option in the Tesla app.
2. Tesla doesn't always reliably apply the configuration changes. Netzero automations work around this by retrying configuration changes when they are not applied successfully. I have not added those retries to the API endpoints yet.

FynnGolfer, 2024-07-04:
Thanks, I will keep trying various options and see if I can solve my problem.Sent from my iPad

closed 2024-11-05 by zigam