netzero-labs/netzero #21: FR: Amber Advanced Price Forecasts - NetZero integration

purcell-lab, 2024-07-06:
Thanks for the work on NetZero being able to inject Amber dynamic pricing into the Powerwall.

This has been working well with the Telsa Time Based Control (TBC), charging the battery during the cheapest times overnight and during the day and then exporting during the peaks in the morning and evenings.  If anything TBC seems to cycle the battery a little too much as it tries to extract the maximum value against a dynamic tariff.  

One other feature of the Australian spot price forecasts is they continue to update/ change even within the 30 minute timesteps, so you might enter a timesteps with a high forecast, TBC starts discharging, but then the prices collapse.  This also tends to have TBC not exporting for a reasonable price as it sometimes waits for a higher price later, that then disappears.  A classic bird in the hand vs two in the bush problem.  In the screen shot below you can see the AEMO forecast at 20:00 has a peak, which TBC would wait for.  Our retailer, Amber, has now produced what are called Advanced Price Forecasts, which are the purple shade and reflective based on experience of what the prices actually are going to look like in the future.

[image]

The good news is these Advanced forecasts are now also available via the Amber API, which NetZero is already accessing, and publishing not only spot prices but expected high and low advanced forecasts.

https://github.com/amberelectric/public-api/discussions/214

It would be great if NetZero was adapted to have an option to choose between injecting the standard AEMO forecasts (as it does now) or these high or low forecasts for Powerwall TBC.  Consensus seems to be the low forecasts are most useful.

```json
[
    {
        "type": "CurrentInterval",
        "date": "2024-07-05",
        "duration": 30,
        "startTime": "2024-07-05T05:30:01Z",
        "endTime": "2024-07-05T06:00:00Z",
        "nemTime": "2024-07-05T16:00:00+10:00",
        "perKwh": 28.94841,
        "renewables": 31.468,
        "spotPerKwh": 13.94436,
        "channelType": "general",
        "spikeStatus": "none",
        "tariffInformation": {
            "period": "offPeak",
            "season": "default",
            "demandWindow": false
        },
        "descriptor": "veryLow",
        "estimate": true,
        "advancedPrice": {
            "low": 29.38334,
            "high": 36.9962
        }
    },
    {
        "type": "CurrentInterval",
        "date": "2024-07-05",
        "duration": 30,
        "startTime": "2024-07-05T05:30:01Z",
        "endTime": "2024-07-05T06:00:00Z",
        "nemTime": "2024-07-05T16:00:00+10:00",
        "perKwh": -13.58623,
        "renewables": 31.468,
        "spotPerKwh": 13.94436,
        "channelType": "feedIn",
        "spikeStatus": "none",
        "descriptor": "high",
        "estimate": true,
        "advancedPrice": {
            "low": -13.98163,
            "high": -20.90241
        }
    }
]
```

rob0101, 2024-07-06:
It might be worth monitoring the success rate of "low" before deploying.
[image]

With my state's prices, "low" was best during the day. Now during peak hours, it's not ideal.

Here' an example.  The "low" was a poor prediction until the last 30m started.

purcell-lab, 2024-07-06:
> It might be worth monitoring the success rate of "low" before deploying.

A couple of different measures of success. 

. How close was the forecast as the time interval approaches - your measurement graphic above is excellent
. How did the forecast modify the peak windows, my observations is that the APF moves the peak left between 30-60 minutes, which means even if the forecast value isn't close an automation can start discharging or taking other actions on the true peak windows 
. How did the forecast modify future high/ excessive prices. Often we have seen optimisations wait and miss exports for a good price now because it can see a great price later (2-12 hours), but when later comes it has actually collapsed into a poor price. Again if the actual forecast isn't realised as long as the relative values are correct it will be useful for automation.

How to visualise those other measures of success is a good question.

rob0101, 2024-07-06:
At the moment, with a small sample to work with, using the average of "high", "low" and "aemo"  gives some consistency of being "least wrong" most of the time.

rob0101, 2024-07-06:
average(low, low, high, aemo) is another interesting one.   It biasses towards "low" but doesn't ignore the other two alternatives.
[image]

zigam, 2024-07-07:
Thanks @purcell-lab for the suggestion and @rob0101 for all the analysis! I'm happy to add the new pricing data, although I'd rather not make AEMO/low/high an option the user has to make (it gets pretty technical and I don't think most users will be able to make an informed decision here). Since the API is new, it might make sense to gather a bit more data before deciding on the best price signal?

rob0101, 2024-07-08:
I'm seeing all three options alone being less than ideal candidates.  
AEMO can have a better success rate but also a worse failure rate. The other two alternate between being best.
I'll come back with a suggestion once I've seen more data.

rob0101, 2024-07-10:
After 95 hours of forecasts...

[image]

Amber's "high" outperforming "low", which isn't a good sign if Amber's Smartshift favours "low".
All 3 choices now have similar rates of success to their own failure rates.

Choosing the middle value each time (the test formula) is still a clear leader.

rob0101, 2024-07-10:
I've change my comparison metric to average cents 'wrong' per prediction.  I think this compares each choice on equal terms, not just wins and last places which ignores coming a close 2nd or close last.

For ~ 100 hours of predictions:
       Amber low:  7.5c,  high: 5.8c,  AEMO 5.1c, middle value: 4.3c

The lower number, the better.   Choosing the middle value is a clear leader still.

The issue with this method is failed spike predictions, where a single bad prediction could be near 1700c 'wrong'.
I might need to put an upper limit on 'wrongness' per prediction, perhaps a 100c cap.

awulf, 2024-07-10:
Is the API documented anywhere to pass these dynamic prices to the PowerWall? Then we could experiment ourselves.

zigam, 2024-07-10:
> The lower number, the better. Choosing the middle value is a clear leader still.

Thanks @rob0101! To confirm, the middle value is `avg(low, high, aemo)`?

zigam, 2024-07-10:
> Is the API documented anywhere to pass these dynamic prices to the PowerWall? Then we could experiment ourselves.

https://developer.tesla.com/docs/fleet-api#time_of_use_settings

rob0101, 2024-07-10:
> > The lower number, the better. Choosing the middle value is a clear leader still.
> 
> Thanks @rob0101! To confirm, the middle value is `avg(low, high, aemo)`?

No,  I mean the number that sits in the middle.     If A > B > C,  choose B.

rob0101, 2024-07-11:
@zigam, please be aware my formula has only been tested against the feed for my state, there are 4 states and one other region, each with their own prices and predictions.

It would be good to see at least one other state's analysis to confirm the pattern.

rob0101, 2024-07-16:
10 days of data for SA.
Average cents away from the final price:  aemo: 3.9c, low: 6.4c, high: 5.8c, median: 3.5c

rob0101, 2024-07-16:
Amber are going to add their preferred predicted price to the API in the next few days.
Guessing it will appear alongside "low" and "high" in the forecasts.

zigam, 2024-07-16:
Thanks for the update. I've added support for optionally using advanced price forecasts. I can turn it on for individual accounts, if you're interested in testing this, just send your Tesla email address to ziga@netzeroapp.io and I'll enable it for your account.

rob0101, 2024-07-17:
"advancedPrice" : {
         "high" : 49.82354,
         "low" : 35.43986,
         "predicted" : 38.73557
      },

I'd say it's safe to switch to this new value rather than AEMO.

zigam, 2024-07-17:
The `predicted` advanced prices are now live for everyone, let me know how it works for you.

rob0101, 2024-07-21:
[image]

These are average cents away from the final price.

The Amber "predicted" price is working much better than AEMO in SA at the moment.
I found a formula that slightly beats it, but not by enough to bother with.

closed 2024-08-06 by zigam

mikepotts64-prog, 2026-09-08:
Is it possible or useful to use Amber dynamic pricing for another wholesale dynamic  provider (LocalVolts) not listed in Netzero? 
Localvolts publishes an API key and provider number.

zigam, 2026-09-08:
> Is it possible or useful to use Amber dynamic pricing for another wholesale dynamic provider (LocalVolts) not listed in Netzero?
> Localvolts publishes an API key and provider number.

We don't support LocalVolts currently, we'll add this to the list of future improvements.

mikepotts64-prog, 2026-09-08:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Thanks Ziga for the prompt response.
NetZero is the most effective controller for my Tesla PW2, gateway setup.

Without API control it remains a blunt instrument particularly in the
negative Feed-in-tarif context.

I look forward to an API solution  being able to make full use of NetZero
function.

regards,
Mike Potts
0427120012
e ***@***.*** ***@***.***>

mikepotts64-prog, 2026-09-15:
Trying to subscribe for a year and get the following error.
Same error occurs on month to month subscription.
Advice please?
Looking forward to being able to sync with localvolts dynamic pricing API
to make effective use of your app.

regards,
Mike Potts
0427120012
e ***@***.*** ***@***.***>

zigam, 2026-09-15:
The screenshot didn't come across, send us an email to feedback@netzero.energy.

mikepotts64-prog, 2026-09-15:
Email sent.

regards,
Mike Potts
0427120012
e ***@***.*** ***@***.***>