netzero-labs/netzero #24: Bob

Bobcraven4, 2024-07-26:
My battery was swapped out several months ago. It has old battery and New battery on diagnostic screen. Still only giving data for old battery.

zigam, 2024-07-27:
Thanks for the report. This is unfortunately an issue with the data coming from Tesla, they doesn't seem to decommission hardware on their backend. I might add an option to manually "mark as decommissioned" specific batteries.

If you connect directly to the gateway (last tab in the Diagnostics screen) you should be able to get a more accurate representation of the batteries, and also more detailed diagnostics per-battery.

zigam, 2024-08-20:
The latest app version will hide battery serial numbers that are not shown in Diagnostics.

closed 2024-08-20 by zigam