netzero-labs/netzero #29: PW2 full 5kW charging

IanPeake, 2024-08-03:
Is there a way to force-charge the PW2 at the full 5kW via the NZ API?

I can set backup_reserve_percent to 100, say, but only get about 1.8kW. This is in self consumption.

While my retailer Amber.com.au has control and charging at 5kW, the NZ API reports nothing remarkable. In particular they don't seem to be manipulating backup reserve. Do they have "special powers" granted by Tesla?:

"backup_reserve_percent":20,"operational_mode":"self_consumption","energy_exports":null,grid_charging":false,"percentage_charged":20.1,"grid_status":"Active (on_grid)","live_status":{"solar_power":418,"percentage_charged":20.080907797564464,"battery_power":-4990,"load_power":684,"grid_status":"Active","grid_power":5256,"generator_power":0,"island_status":"on_grid","storm_mode_active":false,"timestamp":"2024-08-04T07:53:02+10:00","wall_connectors":[]}}

The above is a piecemeal copy-paste from a character terminal without line wrap; forgive me if I accidentally introduced a syntax error that didn't exist in my own logs.

Loving this service; thanks for doing it!

zigam, 2024-08-03:
Netzero unfortunately doesn't have control over the Powerwall charging rate. The logic Tesla uses based on observations:

1. Low rate of ~1.8kW when state of charge is below backup reserve and Powerwall needs to charge up.
2. Higher rate of ~3kW for Storm Watch and VPP events.
3. Highest rate of 5kW when in Time-Based Control mode and Powerwall charges based on the utility rate plan.

Amber uses a VPP API, only available to utilities, which gives them more control over the charging rate (but also restricts them from going below 20% reserve).

IanPeake, 2024-08-03:
Thanks. It looks like NZ supports switching to TBC, but not dynamically setting the rates?

zigam, 2024-08-03:
> Thanks. It looks like NZ supports switching to TBC, but not dynamically setting the rates?

You mean setting the utility rate plan? There is support for that in the app, but only when the rates are dynamic (e.g. Amber Electric, Agile Octopus, Tibber).

rob0101, 2024-08-03:
Amber can set any rate (positive or negative) they want via the vpp api.  I've asked them if we can use that.  No answer.

In the meantime there is Tesla's hidden "backup" mode that will charge at 3.6kW.  That's what I use.

zigam, 2024-08-03:
> In the meantime there is Tesla's hidden "backup" mode that will charge at 3.6kW. That's what I use.

Ah thanks, you suggested this mode in the past. That's undocumented, but a supported mode with Netzero API: https://github.com/netzero-labs/netzero/issues/3

IanPeake, 2024-08-04:
Thanks @rob0101 and @zigam that solves my immediate problem very nicely .

closed 2024-08-06 by zigam