netzero-labs/netzero #38: MRL got stuck

rob0101, 2024-10-15:
Had a weird thing happen overnight.

My automation sent a request to change the MRL (backup_reserve_percent) from 59% to 5% via the Netzero API last night.
`Calling netzero with {"operational_mode":"self_consumption","backup_reserve_percent":5}`

response:

`{"backup_reserve_percent":5,"operational_mode":"self_consumption","energy_exports":null,"grid_charging":false,` .....

So Netzero sent the request and showed a confirmation the request was successful.

Many hours later I find the Powerwall SOC had stayed at 59% the rest of the night.

This morning the Tesla app showed the MRL was indeed 5%, but my Home Assistant monitoring (read only activities) showed the MRL was still 59%.

My feeling is that the Powerwall is in a confused state, giving different answers

I'm curious if others have seen something like this.

BJReplay, 2024-10-16:
Not directly answering your question, but to the extent that a multiply negative response is helpful in any way...

I'm using the Tesla Fleet integration in Home Assistant (using my own Tesla Fleet account oAuth credentials rather than the built in credentials) to change my reserve twice a day to set my Powerwall to either charge during free charging windows or discharge into load.

This has been working well for the last few weeks.

Before that I was using my own automation (again, using my own Telsa Fleet developer account credentials) rather than NetZero.

Hasn't skipped a beat.

I had been - up until recently - using NetZero as the braces in case the belt failed - issuing a "just in case" command five minutes later.

However, I figured I'd ease off the API count for NetZero since I didn't seem to need it.

What has changed - for me - is that since my Powerwall Gateway was upgraded to Firmware: 24.36.1 f997968b a week or so ago, local login and monitoring has been flakey as can be.  Dropouts for hours on end.  Known local passwords not working to log in, and the necessity to reset the local password to log in.

I wonder if you have local monitoring as well, and you're on that firmware, and you're seeing the side-effect of what I suspect is a flakey firmware.

zigam, 2024-10-16:
I've gotten a couple reports of configuration changes getting lost like this. I suspect if the gateway is in a bad state, or in the middle of a firmware update, or there are networking issues, the change can fail to propagate from the cloud to the gateway. The Tesla app and cloud APIs will show the cloud state, which might be inconsistent with the local state when that happens.

I've been adding some redundancy to try to work around this, but Netzero unfortunately can't verify the local state. Since you're running your own automation you might want to run an additional local check, and retry the automation if needed.

zigam, 2024-12-05:
The lost configuration changes seem to caused by intermittent failures somewhere on Tesla's side. Netzero now runs every configuration change twice in short succession, and so far I haven't heard about new cases of automation failures.

I just enabled this for API calls as well, so if you run into this issue again please reopen.

closed 2024-12-05 by zigam