netzero-labs/netzero #45: Support for ComED Hourly Pricing in the USA

petersloan, 2024-10-31:
Hi

I noticed you are now adding support for dynamic pricing schemes.  The major electric provider for the Chicago metro area is ComEd and they have a hourly pricing scheme.  The day is divided into 1 hour blocks and the real time 5min prices are averaged from the top of the hour to the current time.  There is an API call (returns JSON) that does this for you and is updated every 5mins: https://hourlypricing.comed.com/api?type=currenthouraverage

The API is open and requires no auth.  The documentation for the full API is available here: https://hourlypricing.comed.com/hp-api/ 

The 24hr 5 min API is mostly provided for transparency and so trends can be determined so might not have a use in NetZero.

Your consideration is much appreciated

zigam, 2024-11-01:
Thanks for the suggestion, ComEd support is on the roadmap.

The API they offer is actually not very useful, because it's only for historical or current pricing. For Time-Based Control, Powerwall needs **future** pricing information, so it can determine the off-peak and peak periods and charge/discharge accordingly.

The Netzero integration will likely be based on [PJM day-ahead pricing](https://www.gridstatus.io/graph/lmp?iso=pjm), which will approximate real-time pricing for the following day.

More discussion here: [Automatically switch to Powerwall on High Energy Prices](https://www.reddit.com/r/Powerwall/comments/1fjdg5w/automatically_switch_to_powerwall_on_high_energy/).

petersloan, 2024-11-01:
Thank you for this feedback.  I will note that the PJM day ahead pricing is "predictive" and not "prescriptive".  This means this is the anticipated rate and can and will often deviate, sometimes quite wildly, from what the predicted day ahead rates are as the hourly pricing is based on real time grid conditions.  

As I am looking to control based on real time pricing conditions, I will likely need to look into using the netzero API with my own script to monitor real time rates and adjust accordingly as I won't feel comfortable relying on the day ahead pricing without at least a real time price guard in place.  We are no TX but I have seen price swings > $1/kwh during extreme events.

I will go ahead and close this out since there is already another discussion tracking this.  Appreciate all the effort!

zigam, 2024-11-01:
I should have mentioned: the "current hour price" API will also be used to adjust the tariffs. It's just that sometimes these real-time changes can't be acted upon, if the Powerwall had already charged up or depleted based on day-ahead pricing.

Some other dynamic tariff plans that Netzero supports (e.g. [Octopus Agile](https://octopus.energy/smart/agile/) in the UK, [Tibber](https://tibber.com/en) in Europe) use day-ahead prices instead of real-time prices, which I think is a better trade-off -- it allows one to plan ahead and change behavior, while still capturing the swings in supply and demand. But Netzero also supports [Amber Electric](https://www.amber.com.au/) in Australia, which is a real-time pricing plan similar to ComEd Hourly Pricing.