netzero-labs/netzero #47: You do not have permission to perform this action

rob0101, 2024-11-05:
Just started to get this from the API..

Calling netzero with {"operational_mode":"self_consumption","backup_reserve_percent":61}
{"detail":"You do not have permission to perform this action."}

Something broken with Netzero or Tesla playing "remove the feature" ?

BJReplay, 2024-11-05:
Not necessarily answering your question, but:

I am running firmware 24.36.2 46990655 which has broken a bunch of stuff (including home assistant integration using the traditional local integration - it seems to lock up the local login anywhere from 6 to 24 hours after I reset the password, and I have to go through a toggle powerwall off, forgot customer password reset (via 192.168.91.1 route or WiFi connection to gateway) to reset, toggle powerwall on) to regain access.

However, my perhaps-not-helping data point is that setting the Powerwall via the Tesla Fleet integration (exactly what netzero is doing) where I set both operational_mode to self_consumption and backup_reserve_percent (to either force self consumption by setting backup reserve to a low level, or charging by setting operational_mode to autonomous and backup_reserve_percent to 100%) is continuing to work, even though the local gateway is refusing to allow me to monitor things.

So, TL;DR, I am seeing "remove the feature" of local monitoring in the latest gateway firmware, but still see the ability to control reserve and mode via Fleet API.

zigam, 2024-11-05:
@rob0101 Perhaps expired Tesla tokens, if you log in to the Netzero app or web, does that fix the issue?

@BJReplay Do you have a Gateway 1? There [have been reports](https://teslamotorsclub.com/tmc/threads/unable-to-log-into-gateway-after-24-36-2-update-work-around-found.336050/) of a passwords reset in that firmware version, but unclear so far how this affects Gateway 2.

BJReplay, 2024-11-05:
> @BJReplay Do you have a Gateway 1?

Yeah, I do.  I had pretty much the same problem with 24.36.1, and was downgraded back to 23.28.2 for a few weeks (without asking) which makes me think that Tesla knew there was something wrong with 24.36.1.

As that post thread eventually gets to, you seem to:

1. Need to be connected to either TEG-NNN WiFi (or LAN 192.168.91.1 via a route)
2. Turn off **all** Powerwalls if you have more than one, before attempting the rest.

I had four things using the password (including NetZero), and I'm down to two (including NetZero), but even if I'm really careful to make sure that all things are turned off (e.g. not attempting to log in / poll with a bad password) while changing, something seems to cause a lock up.  I suspect they've got something as innocuous as a memory leak in the gateway firmware that causes the lock outs - at least for frequent polling scenarios.

rob0101, 2024-11-05:
I was asked to login again in the app.  It didn't fix the api issue.

zigam, 2024-11-05:
@rob0101 I'd need some more data to investigate, can you send me an email at ziga@netzero.energy?

himynamesned, 2024-11-05:
I was booted early hours too. Able to establish new token and change configuration via NetZero

rob0101, 2024-11-05:
@zigam fixed mine soon after seeing the logs for my issue. 
👏
Normal service has resumed.

closed 2024-11-14 by zigam