netzero-labs/netzero #48: Octopus Intelligent Go

craigmase, 2024-11-09:
Hi first of I all love the app

When I set the utility plan to OIG everything seems to worth fine with the automation but the Powerwall 2 is exporting too early (generally between 3-7pm) to where the SOC drops to low then I'm forced to use the grid which is undoing all the export I've gained during the day. 

Ideally I'd want to retain the SOC until say 9pm then start to export ending up a 0 at 11:30pm

craigmase, 2024-11-09:
[image]

zigam, 2024-11-10:
Thanks for the report. Did the automations help with the problem?

Powerwall behavior in Time-Based Control is determined by Tesla, Netzero just provides the up-to-date tariffs. I'm actually surprised it would export on OIG, I assume you have a fixed export price? It seems avoiding pulling from the grid at the high daily rate would be more important than exporting at a fixed price.

closed 2024-11-22 by zigam