netzero-labs/netzero #49: Seasonal Automations - Feature

farberm, 2024-11-10:
I do not see this option to adjust automations based upon seasons. Is this feature coming in an upgrade? It would be nice so that I do not need to remember to adjust things every spring and fall based upon my electric company (Duke Energy Carolinas). 

Another feature would be the Critical Peak Period which is being oimplemented to handle that scenario.

Thanks for a great add

zigam, 2024-11-10:
Thanks for the suggestion. Is it just the utility rate plan that changes between seasons, or are the other automation changes? Utility rate plans in the Tesla app do have support for seasons currently.

I'll have a look at Critical Peak Period, Netzero would need programmatic access to these periods to be able to act on them.

farberm, 2024-11-10:
The number of periods change as well as the time for each period. In addition M-F is different from Sat, Sun.. I added an image of that is helpful about the changes between season. I do not know how NetZero app does the programming but adding a date range and day of the week would be hrlpful….
[image]

Thanks in advance

zigam, 2024-11-11:
Thanks for the details. Does the Tesla app allow you to configure this rate plan in the app? If so, are you able to use Time-Based Control, or do you use Netzero automations instead to control Powerwall behavior?

farberm, 2024-11-12:
Yes you can configure a custom rate plan and set the seasons based upon months. Yes I can do time based control but no option for automation throughout the day to change things to minimize grid usage during peak time. That is what I use netzero for.. only issue is that I have to reprogram it for each season. If there was also season section to the day of the week area that would be great so it changes automatically based upon the season/montj

zigam, 2024-11-12:
Makes sense, thanks for the additional info!

closed 2026-01-21 by zigam