netzero-labs/netzero #51: Automations not always working

NDGLB03, 2024-11-13:
I use Intelligent Octopus Go and I also get free electric sessions of around two hours each when I can draw all the house can eat.

In order to avoid draing the battery, I use one of two automations (Not at the same time, I've learnt that);
Firstly;
When home usage rises above 6.5 kW: Set backup reserve to: 100%; Set grid charging to: Enabled.
When home usage drops below 6.5 kW: Set backup reserve to: 25%; Set grid charging to: Enabled.

That works when the sun isn't shining. I have solar panels and sometimes their generation drops the demand below 6.5kW and the battery has drained. 

So I set up another option for use during the day;
Every day at 9:00 AM: Set backup reserve to: 100%; Set grid charging to: Enabled.
Every day at 11:00 AM: Set backup reserve to: 25%; Set grid charging to: Enabled.
The 9am and 11am are what it was set to for today. I change the times to match the free electric slots.

Twice now, I've received the "success" e-mail for setting to 100% at a set time but the battery has continued to drain. To fix that I unpaused the home usage rule and that has kicked the battery into charging.

Can I manually set the battery under "configuration"?
Is this a software issue or an issue with my internet?

zigam, 2024-11-14:
Thanks for the report!

> I have solar panels and sometimes their generation drops the demand below 6.5kW and the battery has drained.

 I didn't understand this part, why does solar generation affect demand?

As for the overall problem with automations not kicking in: I've seen some cases of failed automations, unfortunately Tesla is not always reliable at passing down the configuration changes, and there's no way for Netzero to verify that (Tesla makes it seem like everything was successful).

One thing that helps with reliability is keeping the actions simpler. In your case, the "Set grid charging" action is unnecessary, because it never changes (and also Tesla started ignoring this option).

Finally, you can manually set the backup reserve under Configuration, yes.

NDGLB03, 2024-11-14:
Thanks for the reply. 
To answer your question;
My car charger is a Zappi (I missed that bit out) so the Powerwall only sees a house demand. When the sun is shining, my grid demand can drop below 6.5kW which, I discovered, activated the second rule and caused my battery to drain. 

I wasn't aware of the "set grid charging" point. Thanks for that.

himynamesned, 2024-11-14:
@NDGLB03 FYI OE now allow Agile export on Go tariff and NetZero supports this this combo now.. I made the switch last week. It's now cheaper for me to charge at the overnight rate and allow export surplus to grid during the day. 

Powerwall also max discharges during peak export rates to make use of max spread - lowering my bills. I now just leave in Time Based Control and Stop mode with zappi during the day to maximise savings. 

I just now need an automation to toggle the zappi between modes for certain time of days. 

Gone are the days of maximising use of surplus solar as it's cheaper for me to export as usually above the Go cheap rate.

For free electricity sessions, I'll likely just override the NetZero tariff temporarily or change max backup to 100% via automation - and dump into grid prior to make room. 

Maybe food for thought...?

OE allowed me to switch export tariff within 5 mins of sending them a direct message on X

NDGLB03, 2024-11-15:
@himynamesned I'm currently in the process of opting out of the FIT export rate, adding my new solar panels and transferring to the SEG. Because of all that, I'm currently staying with Scottish Power to keep things simple. After that, I may transfer to OE.

NDGLB03, 2024-11-17:
Interesting delay this afternoon. I plugged my car in at 15:03 and IOG came up with a plan starting then.
Netzero reported the automation worked at 15:05 but the Powerwall was discharging. The Tesla app confirmed that the backup reserve was set to 100% but the battery was still discharging.
It wasn't until 15:18 that the battery started taking a charge.
It wasn't a problem as the battery was almost 100% charged but is there any idea why it took so long to kick in?

zigam, 2024-11-18:
> Interesting delay this afternoon. I plugged my car in at 15:03 and IOG came up with a plan starting then. Netzero reported the automation worked at 15:05 but the Powerwall was discharging. The Tesla [...]

Not sure why the delay, could be an issue on Tesla's backends. I'm going to spend some more time investigating the reliability of automation commands, how long they take to get passed down, and how often they fail.

Adrius42, 2024-11-21:
My issue doesn't quite fit this topic:  It's more switching OFF Automation does not behave as I expected. When I switch off automation, I expect the state of the system to be returned to normal meaning control is returned to the normal controller in normal state.

My automation involved switching off Grid Charging despite Timed Charging based on Octopus Times being set in the Gateway, it delays charging until 1:00 am allowing my car to take Full Incoming Amps. It also limits charging to 95%.

My goal was to allow 100% battery charging and to allow grid charging from the normal Gateway Timing.

I simply switched off the Automations.
Sadly as automation had Switched Off the Grid Charging at the time I did this, the result was a completely flat set of Power walls the following morning. Darn it!!

My expectation was Switching Off Automation would return the system to the state it was in before I applied Automation.

At the very least a warning of Grid Charging Status would be useful.

Adrius42, 2024-11-21:
@hymynamesned  Good luck getting Scottish Power to switch off FIT Export!!!  My only suggestion is if you have an electric car. Switch NOW! 7p charging for your car and battery! Versus soon to 35p! As an incentive if you use this link you can share £100 with me! Your Octopus Energy Referral code
Use this link to sign up to Octopus - https://share.octopus.energy/open-run-414

Seriously, I have been trying to get switch off FIT Deemed Export via Scottish Power since July!! Still at it, they are very effective at prevaricating and REFUSE to resolve issues over the phone!

That's a lot of 28p per KwH saving missed!!!I've used over a MW from my batteries since I started trying to switch off deemed export!

zigam, 2024-11-22:
> Imagine if my Tesla was braking when I switched off automation and the car continued braking!!!

I responded to your email, I would need some more information to understand the issue.

zigam, 2024-11-22:
Netzero will now run automations twice for improved reliability, so I'm closing this for now. Please reopen if any automations fail again.

closed 2024-11-22 by zigam