netzero-labs/netzero #52: Automation did not happen

Kat-A-Tonic68, 2024-11-15:
For some reason I checked my powerwall app this morning and found that I was drawing from the grid when I should have been running the house on the powerwall.  Netzero app said that the automation was successful, yet it had not changed the setting.  I pay horrible demand fees for usage during peak times, and have found through some searching that occasionally the signal from netzero goes to Tesla and is shown as completed, but the signal does not get from Tesla to me.  Is it OK to have netzero double up on comands say 5 minutes before an event to make sure it happens?  I am talking about having it set powerwall reserve to say 15% at 4:55 then to 10% at 5:00 just to make sure it happens?

zigam, 2024-11-15:
Hi, thanks for the report. Yes, ok to to add a "backup" automation like that. I'm considering doing something like this in Netzero automatically, since I've gotten more reports recently of automations failing to be passed down by Tesla.

Kat-A-Tonic68, 2024-11-15:
Thanks for the response.  I will add them.

zigam, 2024-11-20:
Netzero now runs automations twice in short succession, let me know if you're seeing more reliable runs.

NDGLB03, 2024-11-20:
> Netzero now runs automations twice in short succession, let me know if you're seeing more reliable runs.

I've set mine to send me e-mails when an automation is activated. Just so I can expect it, will I now get two e-mails for each automation?

Kat-A-Tonic68, 2024-11-20:
I had only had the issue one time, and after I reached out I had it set
upto run it twice 5 minutes apart.  No issues since then.  I will let you
know if it misses one in the future.

Thanks!

zigam, 2024-11-20:
> I've set mine to send me e-mails when an automation is activated. Just so I can expect it, will I now get two e-mails for each automation?

No, single notification. The retry happens behind the scenes.

zigam, 2024-11-21:
> I had only had the issue one time, and after I reached out I had it set upto run it twice 5 minutes apart. No issues since then. I will let you know if it misses one in the future. Thanks!

Thanks! And if things are running reliably, please also test with removing the second automation. I'd like to make sure the automatic retry is working well enough that the second automation is not needed.

NDGLB03, 2024-11-21:
I've got a timed automation for tonight with just the one automation. I'll also report back.

NDGLB03, 2024-11-22:
I've now executed a number of automations - all single instructions from my end and I'm happy to report that they all worked fine.

Many thanks.

zigam, 2024-11-22:
> I've now executed a number of automations - all single instructions from my end and I'm happy to report that they all worked fine.

Great, thanks for the update! I'll close this issue for now, but please reopen if it happens again.

closed 2024-11-22 by zigam