netzero-labs/netzero #54: Charging Powerwall during IOG extra session

AndyP19121965, 2024-11-19:
Hi,

I’m new to Netzero and cannot find a manual.

I'm on the IOG tariff. I would like to use automation to manually set the Powerwall to charge during the extra charging sessions e.g. I’ve plugged my car in and IOG will charge the car from 15:00 to 15:30, I would like to charge the battery during this period.  Can I do this?

NDGLB03, 2024-11-20:
I do it is as follows;
When home usage rises above 6.5 kW: Set backup reserve to: 100%.
When home usage drops below 4.5 kW: Set backup reserve to: 25%.

If you cook electric (I don't - yet) you may find the house demand could trigger the first rule so you would need to tailor it to your needs. What you can also do is pause the automation and restart it when you have an extra charging session. 

What I've also found is that IOG may start a session and then pause it due to demand elsewhere. This takes account of that and stops charging the battery at what will become (I think) an expensive rate.

AndyP19121965, 2024-11-20:
Many thanks.My PW does not see my car charging to avoid battery drain and I have an ASHP, so I wouldn’t want the battery to charge in response to the heating going on.However, I could manually set the battery reserve to 100% for the additional IOG periods.  Can I just check, when battery reserve is set to 100%, the PW will charge the PWs battery, regardless of the tariff configured within the PW?Many thanks for your help,Andy

NDGLB03, 2024-11-20:
When I set my reserve to 100%, the battery charges regadless of the tariff.

I have a couple of standby automations which spend most of their time paused but which might suit you better. They are;
Every day at 9:00 AM: Set backup reserve to: 100%.
Every day at 11:00 AM: Set backup reserve to: 25%.

This is for when I get a "Power up" session (Free electric) and the car either isn't home or is already charged.

I forgot to say last time, on the NetZero website, go to the "automations" page, scroll to the bottom and you will find "Automation Documentation".

AndyP19121965, 2024-11-21:
Many thanks, that worked.It’s a shame there isn’t integration with IOG to trigger based on the additional charging slots with reduced tariff.CheersAndy

AndyP19121965, 2024-11-21:
I’ve just noticed that the app has replaced my tariff in the app.  I’m not on agile so wanted to keep the tariff settings I had.  After I replaced it yesterday, the app has changed it again.  How can I ensure the tariff I define in the Tesla app is not replaced?ThanksAndy

The IOG charging slots are unfortunately not available in the public Octopus Energy API. We may add this in the future, but it would require you to share your Octopus API key with Netzero.

closed 2024-11-22 by zigam