netzero-labs/netzero #58: Prioritise solar to Zappi before PW3

ZappiPW3, 2024-11-27:
My single phase Zappi (EV charge point) is installed upstream of the Powerwall 3 (PW3) CT so the PW3 does not see it as a load 
 and therefore does not discharge to the EV
BUT the PW3 thinks that the Zappi is export, not a load, so if PW3 is not fully charged it takes all excess solar output so Zappi gets none.
PW3 is DC connected to solar plus there are two other inverters connected to switchboard that PW3 can see.
Zappi is set to use solar during the day and then boost at night to bring EV up to 80%

How can I configure DC connected PW3 and Zappi so that:
Zappi gets first use of solar up to 7kW.
PW3 does not discharge through Zappi to EV ever.

Thanks

ZappiPW3, 2024-11-28:
> Tesla unfortunately doesn't offer much control over the Powerwall, so changing the order of what gets charged by solar is not something Netzero can do.
>
> Netzero can however avoid the Powerwall discharge into the EV battery. There's is currently an integration for Tesla Wall Connectors and one based on house usage. We're also adding integrations soon for all Tesla cars and for OCPP-compatible EV charges (which includes Zappi).

Hi,

Thanks for the reply. I look forward to the Zappi integration into Netzero. Once it is released I will move the Zappi back downstream so the PW3 can see it as a load.

Hopefully then the EV (Volkswagen ID.3 and Honda E) will get priority on solar production during the day but the PW3 won’t discharge into the EV at night.

purcell-lab, 2024-12-01:
There are two use cases you are after in this request. 

1. Not having the Powerwall discharge when charging your EV. This can be achieved by setting the back up reserve % on your Powerwall to match the current state of charge %. This then prevents the Powerwall discharging. You can either change the backup reserve manually or use a tool like NetZero to Automate.

2. Stopping Powerwall from using excess solar for charging. This is more complex and unfortunately Tesla don't provide a simple 'Preserve' mode. The workaround I use is to force the Powerwall into discharge mode using the TBC hack (by creating an artificial high sell price in the utility rate plan, if Powerwall is exporting it can't charge. Then limit those exports by switching the 'allowed_exports' flag to either 'Never' or 'Solar Only'

zigam, 2024-12-01:
> 2\. Stopping Powerwall from using excess solar for charging. This is more complex and unfortunately Tesla don't provide a simple 'Preserve' mode.

Wouldn't Self-Powered mode accomplish this as well, prioritizing house+EV usage over charging the Powerwall?

ZappiPW3, 2024-12-02:
Currently EV charge point is connected upstream of the PW3 CT so PW3 sees it as an export , not a load. This prevents PW3 discharging into car at night but it also means EV doesn’t get power from solar until PW3 is fully charged (or is getting more than 5kW from external inverters).

I could move EV charge point so that PW3 sees it as a load and use Netzero to stop it discharging to EV at night but I have several large loads that can run at night so that could be tricky.

If and when Netzero can talk to Zappi, I think all of this would be solved.

I am adding more solar PV capacity, with some east facing, so PW3 should be fully charged earlier in the day and issue with EV not charging from solar will be minimised.

zigam, 2024-12-05:
New EV charging automations are now available for testing:
- On iPhone, you can [join TestFlight](https://testflight.apple.com/join/G1nXW3Rp).
- On Android, you can join the beta program from the [app page](https://play.google.com/store/apps/details?id=com.zigam.pwmon&hl=en_US).
- On web, the changes are live.

BJReplay, 2024-12-05:
Pick up my EV tomorrow. Now just need a charger!

closed 2024-12-07 by zigam