netzero-labs/netzero #59: Überwachung und Diagose Gateway

pezet1910, 2024-11-30:
Hallo, die App funktioniert bei mir sehr gut bis auf ein Problem. Wenn ich bei „Überwachung und Diagnose“ die Gateway auswähle bekomme ich ich eine Fehlermeldung: „Failed to authenticate with Gateway/Powerwall+“ Die IP Adresse ist richtig. Das Passwort die letzten fünf Zeichen des Gateway Passworts.
Was muss ich verändern um die Verbindung herzustellen?

Ich benutze:
- Backup Gateway 2
- iPhone 16 Pro Max
- Unifi Accespoints

Gruß
Peter
[image]
[image]

zigam, 2024-11-30:
Hallo,

Die IP-Adressen auf den beiden Screenshots sind unterschiedlich, daher möchte ich sicherstellen, dass die richtige IP-Adresse eingegeben wurde. Netzero kann die IP-Adresse automatisch erkennen, wenn Sie auf die Schaltfläche „Zurücksetzen“ tippen.

Falls die IP-Adresse korrekt ist, können Sie das Passwort mit einem Webbrowser auf demselben Gerät und Netzwerk wie Netzero überprüfen. Öffnen Sie den Webbrowser und gehen Sie zu https://192.168.1.73/ (ersetzen Sie die Adresse ggf. mit der korrekten IP-Adresse). Ignorieren Sie die Sicherheitswarnungen, tippen Sie unten auf „Weiter“, scrollen Sie nach unten und tippen Sie auf „Upgrade später“. Melden Sie sich als Kunde mit Ihrer Tesla-E-Mail-Adresse und demselben Passwort an, das Sie auch in Netzero verwenden. Falls das Passwort nicht funktioniert, müssen Sie es möglicherweise ändern. Das können Sie direkt auf dieser Seite tun.

Ich hoffe, das hilft!

pezet1910, 2024-12-01:
Hallo,
danke für deine Antwort.
Ja, die IP Adressen sind unterschiedlich weil die Gateway zwei Verbindungen hat, LAN und WLAN. Deswegen habe ich auch beide Verbindungen ausprobiert.
Leider habe ich nach wie vor das Problem. Tatsächlich (so wie du schreibst) funktioniert das Passwort nicht. Alle versuche das Passwort zu ändern sind leider gescheitert. Die Vorgehensweise um das Passwort zurückzusetzen ist mir nicht ganz klar. 
- welche Verbindung soll man benutzen, LAN oder WLAN?
- was muss aus und eingeschaltet werden, die Powerwall oder die Gateway?
- was muss dann als Passwort eingetragen werden? Tesla schreibt einmal "die letzten 5 Zeichen der Gateway-Seriennummer" und dann "letzten 5 Zeichen des Passworts" (s. Screenshot)

Gruß
Peter

[image]

zigam, 2024-12-01:
- Sie können entweder LAN oder WLAN verwenden.
- Sie müssen den Schalter an der Seite der Powerwall umlegen. Falls Sie mehrere Powerwalls haben: Einige Nutzer haben berichtet, dass dies für alle Powerwalls gemacht werden muss, aber ich würde es zuerst mit nur einer versuchen.
- Das Standardpasswort für Gateway 2 sind die letzten 5 Zeichen des Passworts auf dem Etikett. Für Gateway 1 sind es die letzten 5 Zeichen der Seriennummer.
- Hinweis: Einige Nutzer mit aktueller Firmware haben auch erwähnt, dass das neue Passwort etwas stärker sein muss, d.h. 8 Zeichen mit Zahlen und Symbolen.