netzero-labs/netzero #61: Powerwall 3Monitoring

rposeyca, 2024-12-02:
Connected to Powerwall 3 wi-fi and connected.  Could not find Strings tab to see strings status?  Where is the Strings tab.?   The Powerwall monitoring shows 3 items; Powerwall, Gateway and Network.
Thanks…

zigam, 2024-12-03:
The Strings tab would appear if data was successfully retrieved, but looks like something went wrong in your case. Do you see the battery metrics like full pack capacity? Are you able to connect using the Tesla One app to see the metrics?

rposeyca, 2024-12-03:
Below is all the data I see on Netzero Monitoring data.The Tesla One app says it’s for employees and External Partners, so I have not been able to access.  Wish I could, my installer could see the power being generated by string.  This is what I was looking for….I use the basic Tesla app.   Here’s the monitoring data.  Suggestions?Sent from my iPad use the Tesla app for monitoring.  Rick Posey

zigam, 2024-12-03:
You can use the Tesla One app, just sign in with your Tesla account under External Partners. Then use the More menu, Tesla Device Setup, and scan the QR code or enter the password.

rposeyca, 2024-12-03:
Thank you, thank you, thank you.,It worked….Sent from my iPadRick Posey

closed 2024-12-05 by zigam