netzero-labs/netzero #64: OCPP with a Zappi

bigalp99, 2024-12-07:
Trying to add my Zappi with OCPP.  I left the Zappi username blank as suggested by Myenergi and now in Netzero - Vehicle Charging, I am getting the following: "Error: Cannot read properties of null (reading 'status')"

zigam, 2024-12-07:
Thanks for the report, this should be fixed now. Already fixed on web, and if you're running a beta app just update it.
It should show "unknown" for the status, to get the status to update just plug the car in temporarily.

bigalp99, 2024-12-07:
Hi 

Thanks for this.

Yes web works now and shows “Plugged in” and “Not plugged in”

PS Myenergi wants a forward slash after the web address to work as shown on their help page

 https://support.myenergi.com/hc/en-gb/articles/16864772981137-Setting-up-Open-Charge-Point-Protocol-OCPP#:~:text=To%20use%20OCPP%2C%20with%20zappi,zappi%20will%20starting%20a%202)

Alan Pearce

zigam, 2024-12-07:
Thanks for the heads-up, we'll update the documentation with specific instructions for Zappi.

closed 2024-12-07 by zigam