netzero-labs/netzero #67: Home graph doesn't display negative power use

18107, 2025-01-03:
When using dispatching from a V2G charger or a second battery, the home should register as having a negative power draw. The graph cuts off all values below 0.

[image]
Home power use from NetZero

[image]
Home power use from Home Assistant

zigam, 2025-01-03:
Thanks for the report. What does the Tesla app show in this case? Netzero should be showing the same data as the Tesla app, since that's where the data is coming from.

18107, 2025-01-04:
The local Tesla connection shows negative home power.
The app previously showed negative home as solar but now shows negative home as positive home power.
The graph shows negative power as zero.

[image]

1.6kW from solar, 3.0kW from home (not to, as the app displays), 4.6kW to grid.

closed 2025-01-04 by 18107

zigam, 2025-01-05:
Thanks for confirming!