netzero-labs/netzero #72: Can’t connect to Powerwall 3

fredct42, 2025-01-27:
In the Netzero Energy app, when I go to Monitoring -> Connect, standing next to the Powerwall, I enter the SSID and password for the powerwall’s network but I get an error “Failure: Error: Cannot detect SSID”.

I’ve tried both Powerwalls I have and I get the same. It’s the same whether I manually type it or use the QR code.

If I go to my iOS settings, WiFi, I can see and join both networks no problem.

However, even when standing right next to them, they seem to be a little intermittent. It seems like, especially after I try to join them in the Netzero app, they sometimes disappear from the Wi-Fi screen in my settings until I turn Wi-Fi off and back on. It seems pretty random though so I can’t nail down a pattern for sure.

Help?

zigam, 2025-01-27:
Can you confirm that iOS location permissions are enabled for the Netzero app? iOS Settings > Apps > Netzero.
(iOS requires location permission for wifi access, but the location is not used by Netzero.)

fredct42, 2025-01-28:
So that was it, or mostly. I had actually allowed location but I had only turned off “precise” location. But I guess that matters. Because turning back on worked.

I still got a few errors complaining about problems joining and some crashes but after a few tries it connected.

One more question. It’s night and the info says all the strings are “disabled”. Is that what it says when there’s no power at night? Or does that mean I connected to the wrong Powerwall? (I wasn’t certain which one was primary. I took an educated guess but could be wrong).

zigam, 2025-01-28:
Thanks for confirming! We'll improve the error message to make it clearer that the location permission is required. We also made some wifi connection reliability improvements in the most recent version.

String status: strings will show Disabled when there's no production, you need to look at the string diagnostics with the sun up.

Finding the leader PW3: the serial number of the leader PW3 should match the serial number of the "gateway" shown in Netzero diagnostics.

closed 2025-02-03 by zigam