netzero-labs/netzero #76: No ability to edit / delete automations

BJReplay, 2025-03-07:
Hi,

Not sure if this is an issue, or I have imagined that you could edit automations and delete automations, but I have lost the ability (if I ever had it) to edit an automation or delete an automation.

I _can_ **create** a new automation, and **Pause all** or **Resume all**.

As I want to enable only some of my existing automations, I am stuck: I can only Resume all or Pause all. 

I have attempted to edit an automation by clicking on the description text, and the icon. 

v2.0.29

BJReplay, 2025-03-07:
Thanks @nickz98!

Would you believe that I tried dragging right, but obviously did not (successfully) try dragging left.

zigam, 2025-03-07:
There is a little prompt at the bottom of the screen indicating that you can swipe (only shown before you swipe for the first time). We'll make that more visible or perhaps keep it on at all times.