netzero-labs/netzero #77: List of available automation commands

Uncle-Paul, 2025-03-07:
Please provide a list of available commands and queries for Netzero automations.

zigam, 2025-03-07:
All of the automation triggers and actions are all listed on the Automation screen (screenshots below). You can also find automation examples here:
https://docs.netzero.energy/docs/tesla/Automation

[image]
[image]
[image]
[image]
[image]
[image]

Uncle-Paul, 2025-03-08:
> zigam
>  left a comment
> (netzero-labs/netzero#77)
> All of the automation triggers and actions are all listed on the Automation screen (screenshots below). You can also find automation examples here:
> https://docs.netzero.energy/docs/tesla/Automation
>
> 1-time.png (view on web) 2-powerwall.png (view on web) 3-ev.png (view on web) 4-usage.png (view on web) 5-price.png (view on web) 6-actions.png (view on web)

Thanks Zigam,

Most helpful, l didn’t know whether there were others documented elsewhere.

I’m trying to sort EVERYTHING out charging with an EO Smart Mini 2 charger that is non -OCPP compatible.

Best regards,

Paul.

zigam, 2025-03-08:
I should mention: if you're in the UK and on the Intelligent Octopus Go tariff, there's also an integration for that:
https://docs.netzero.energy/docs/tariffs/Intelligent-Octopus-Go

closed 2025-03-29 by zigam