netzero-labs/netzero #79: Automate Intelligent Octopus Go - did not restore backup value at end of slot

leolabcat, 2025-03-10:
I tried the new ‘Automate Intelligent Octopus Go’ option with the Octopus API, but it doesn’t seem to be working as reliably as the ‘when charging starts’ automation I was using previously. From last night:

> 22:00-22:30 - Netzero set backup reserve to 100% at start, and restored to 20% at end
> 23:00-23:30 - Netzero set backup reserve to 100% at start, but did not restore at end (stayed at 100% when slot ended, Powerwall log shows that Netzero made a change at start of the slot but not at end)
> 23:44 - I manually changed the reserve back to 20%; car was not charging and this was not in the time slot showing in the Octopus Intelligent Go charging schedule
> 00:30-01:30 - car charged in this Intelligent slot, backup reserve not changed during this slot
> …
> 06:00-06:30 - car charged in this Intelligent slot, backup reserve not changed during this slot

I was concerned when I saw the reserve stay at 100% outside a charging slot (even though this was after 11.30pm) since, without a subsequent change, the Powerwall would not power the home after 5.30am.

Is this expected behaviour for Netzero - i.e. did I intervene unnecessarily and would it have restored the reserve value at 5.30am?

zigam, 2025-03-10:
It would have restored at 05:30. We may make this configurable, but if a charging session ends during the off-peak period, the IOG integration will keep the backup reserve at 100% (since the lower prices persist throughout this period). We added this to the documentation.

leolabcat, 2025-03-10:
Ah, excellent. Many thanks for explaining.

closed 2025-03-10 by leolabcat