netzero-labs/netzero #81: Unexpected Backup Reserve Settings Behaviour

FBurrage, 2025-03-23:
IOG provided a slot from 21:30 to 22:00 tonight.  At 22:12 I noticed that the PW was still charging.  I noted that Netzero hadn’t moved the set point back from 100% to 5%.  I’ve seen this behaviour a couple of times in the last few charges.

I also noticed that there were multiple settings changes recorded by the Tesla app (see attached photo).  It’s not clear what Netzero did at either of the 21:30, 21:33 or 22:00 timestamps.  I’d anticipate that Netzero would only make a single change for each on or off switch.

There also seems to have been several 3 minute offsets to the backup reserve setting changes at various times.  I’d anticipate that changes would only occur to line up with IOG sessions.

Note that I’d made several changes to the backup reserve earlier today so we could get a Bump Charge, but in the end reverted back as we realised we didn’t need the charge.

[image]

Hope this feedback is useful, the app is exactly what we need to make the most of IOG.

FBurrage, 2025-03-24:
Useful to know, thanks for the explanation.  Does the poor reliability of Tesla passing the config changes down to the PW also explain why I’ve occasionally had to reset the backup reserve myself, as happened yesterday?

zigam, 2025-03-24:
It's more likely that there was a scheduled charging session, but to confirm that you'd need to send us an email to support@netzero.energy with your Tesla account email address.