netzero-labs/netzero #84: Multiple Triggers Automation

benzoic9723, 2025-04-07:
Hello, is it possible to have multiple triggers that must be true for an automation to run? For example, I want to have my system to export everything (Solar and Powerwall) right before the end of my peak hours. 

Something like:
If Weekdays after 8pm _**and**_ Powerwall State of Charge is **above** 25%, change exports to Everything.

zigam, 2025-04-08:
This is not currently possible using just the app (other than time restrictions for event-based automations, such as:
`When Powerwall is discharged down to 25% (between 8:00 PM and 9:00 PM): Set energy exports to: Everything (solar and battery)`.

You can use the [Netzero API](https://docs.netzero.energy/docs/tesla/API) for more complicated rules.

benzoic9723, 2025-04-09:
Thanks for the feedback :)

One more quick question, for Powerwall automation : "When Powerwall state of charge is discharged down to: ##% and only between 8-9pm' - does that mean it will activate only at ##% or anything below ##% during that hour slot?

zigam, 2025-04-09:
The former -- the automation would only activate if Powerwall drops below 25% between 8-9pm.

closed 2025-04-12 by zigam