netzero-labs/netzero #85: Request: PW automation condition

jeffcloherty, 2025-04-07:
If would be awesome if a greater than / less than condition could be added to the automations based on the PowerWall charge %.

This would be useful for creating an automation for cloudy days where I can set the operational mode to Time-Based only if the current PW charge is a specific amount below the backup reserve %.

Use case example: on sunny days my PW charge is usually ~60% by noon from solar charging. On cloudy days if the PW charge is below 30% at noon, I'd like to charge from the grid at the TBC rate so they're ready to power the house during higher prices between 3pm-midnight.

UI element example;
New Time Automation
When: [Time of day]
Optional: If PowerWall charge state is: [Below %, Above %, Between x% and y%]
Choose one or more actions:

New PowerWall Automation
When PowerWall state of charge is: [Charged up to %, Discharged down to %, Discharged down to backup reserved, Below %, Above %, Between x% and y%]
Optional: Only between the hours of []
Choose one or more actions:

This app has been amazing so far, thanks for the great product!

zigam, 2025-04-08:
We'll consider more complicated automation rules, in the meantime there's the [Netzero API](https://docs.netzero.energy/docs/tesla/API).

I'm curious, does Time-Based Control mode not work well for you? It predicts solar generation and charges from the grid as needed.

jeffcloherty, 2025-04-08:
Thanks for considering it. I'm on NEM3 so anything sent back to the grid is wasted energy from my perspective. In my experience Time-Based Control would over-charge the PW then send solar to the grid at peak times, my set of rules attempts to make sure grid charging + solar generation approaches but doesn't go over the PW capacity.

closed 2025-04-12 by zigam