netzero-labs/netzero #88: Dynamic Tariff Updates Not Reflected When Price is Negative

petersloan, 2025-04-12:
I have noticed that during periods of negative rates, actual rate is not showing up as the energy rate on the NetZero dashboard.  I can confirm what I see on the dashboard is also what the Powerwall sees in the Tesla app.  Is it possible that the currenthouraverage is not being properly sent over and instead its only sending the day ahead prices?  In this case, on TBC, the powerwalls were exporting at -.10 so I was actually paying .02 to export power until I caught it.

Is there a place where rate updates are logged so users can see what was updated and when?

[image]

[image]

[image]

[image]

zigam, 2025-04-12:
There are no known issues with ComEd prices, but if you send an email to support@netzero.energy with your Tesla email address, we can check your account's data.

Note that Netzero updates prices every 5 minutes, so it's possible that that the real-time price differs from the configured one. Also, Tesla's Time-Based Control does not support negative prices. You could try an automation that turns off exports when prices drop below a certain level (which should reduce solar production to match just your house usage).

petersloan, 2025-04-12:
I did not realize there is a hard floor of 0 cents. By automation, you mean writing something myself using the API?  I am not aware of a conditional I can set other than time based with NetZero.    

Today was a weird situation where prices have been negative all day and very strong solar day, I ended up filling my powerwalls, and then started exporting again.  I couldn't go off grid because the batteries were above 90%.  Setting exports off also did not work as I don't think the Powerwalls (PW3) would raise the frequency to trip the enphase micros unless it was in an off grid situation.  In a one off situation, its not a big deal but power prices have been wild here this year so likely there will be additional days like this where heating/cooling demand is low and because of the weekend, demand is also weak. 

Has anyone else had to handle situations like this and if so what sort of logic did folks end up coming up with.

zigam, 2025-04-13:
You can create event-based automations in Netzero, have a look at the tabs at the top of the New Automation screen.

Negative export prices are not very common, and unfortunately Tesla doesn't support them very well. Most people will go off-grid (manually, since it cannot be automated). In some cases, the "No exports" Netzero configuration works by limiting solar production.

closed 2025-04-13 by petersloan

zigam, 2025-04-13:
I should have clarified -- negative pricing is common for plans with wholesale pricing, but plans with wholesale pricing are still rare, globally.

Good point about frequency shifting and grid following.