netzero-labs/netzero #89: Can't disable utility rate plan during issues with Amber

rob0101, 2025-04-19:
Amber are having some issue with their pricing API.

I went to switch Netzero off, so it doesn't use up my API call allowance, but I cannot switch it off.
Instead of the toggle I get a 502 Server Error message instead.

zigam, 2025-04-19:
Thanks for the report. I can't reproduce this issue now, looks like their API is back. Was the `Clear` button available in `Settings > Utility Rate Plan`? That will turn off updates as well.

rob0101, 2025-04-20:
Clear was available and what I resorted to. 

Many people were hitting API limits due to the issue, so being able to turn off some calls would have been nice. 

It'd be nice to have an API option to turn off the utility rate plan price fetching.

zigam, 2025-04-20:
We'll consider this, you can also switch to Self-Powered mode which will stop the tariff updates.

rob0101, 2025-04-21:
I wasn't aware that TBC <-> Self-consume toggled the tariff updates too.   That's handy.

Do the rates get refreshed just before or after TBC is turned on ?.     With Amber, you'd want the latest prices installed just before enabling TBC, otherwise it might buy during a good sell period.

zigam, 2025-04-21:
Switching modes is independent from the tariffs (and might be done from the Tesla app), so tariffs will be updated after.

closed 2025-05-10 by zigam