netzero-labs/netzero #91: How to force Powerwall 3 to export battery to grid at 12:30am.

ADOS42, 2025-05-10:
Hi
I have a Powerwall 3 and solar system and export permission. I am trying to get the battery to export all but the reserve at 12:30. Then at 3am charge the battery to 100% using cheap rate electricity.
I have set the system to 

Every day at 12:30 AM: Set backup reserve to: 14%; Set operational mode to: Time-Based Control; Set energy exports to: Everything (solar and battery).

But the system does not seem to export power at all.
Any advice on how to do this?
Thanks

zigam, 2025-05-10:
There is no easy way to force Powerwall to export to the grid unfortunately. The Netzero configuration you have is necessary but not sufficient to have it export. You'd need to also modify the utility rate plan to make that period look like peak period (high import and export rates). And there would have to be enough capacity to get you through that peak period without importing from the grid later.

closed 2025-05-15 by zigam

ADOS42, 2025-05-15:
That doesn't work, I tried setting to high export rate. No effect