netzero-labs/netzero #92: OCPP not working with NetZero (Wallboc Commander 2)

aguilaair, 2025-05-13:
Hello,

When setting the OCPP configuration in NetZero, it gets stuck in the "Charging Finished" state, and the charge cannot be started until I disable the OCPP on the Wallbox. I have used EVCC successfully with this same charger so it is presumably an issue with NetZero's OCPP implementation.

zigam, 2025-05-13:
Thanks for the report. A few questions to troubleshoot this:

1. When you configure Netzero as the OCPP provider, does the charger show up with a green light in the the Wallbox app? It should say something like "Plug in your vehicle".
2. If you unplug the vehicle, does the correct state get shown in Settings > Vehicle Charging?
3. If you plug the vehicle back in, does it start charging automatically?

aguilaair, 2025-05-13:
> Thanks for the report. A few questions to troubleshoot this:
>
> 1. When you configure Netzero as the OCPP provider, does the charger show up with a green light in the the Wallbox app? It should say [...]

1. When OCPP is configured it indicates OCPP connected, and the title on the app also shows "Charge Finished"

2. Yes, it shows "Not Plugged In" on the app.

3. The charge does not start, it just says charging finished and does not have the start charge button

zigam, 2025-05-13:
That's an odd issue. The status of the charger is not determined by Netzero, it just gets the updates from the charger. And since the Wallbox app shows the same status, I wonder if the issue is with the charger, connection, or the EV?

We have Wallbox Pulsar Plus chargers running without issues, and many other EV charger brands as well.

aguilaair, 2025-05-14:
So I have figured out the issue, I think? I got it working at least for charging last night. It looks like when the wallbox first connects to OCPP, it cannot be plugged into a car that is already charged? Otherwise, it will get stuck there. I will investigate a little more to see if I can reproduce the issue following a specific set of steps. (I tried OCPP months ago with netZero and moved away because of this same issue)

closed 2025-05-14 by aguilaair