netzero-labs/netzero #93: Multi-system views/switching - Feature request

pmeenan, 2025-05-14:
For houses with more than one service panel, it is not unusual to have multiple powerwall systems running in parallel (one for each panel). The netzero app supports this well and allows for switching between systems through the settings.

It would be helpful if there was a faster way to switch on the mobile app (maybe change the title to a dropdown) and if it was possible to display all of the systems concurrently in the web app (or on a tablet with the native app).

pmeenan, 2025-05-14:
Sorry, just realized that you can switch in the app by tapping on the label so it's just the multi-view web view that would be a useful addition.

zigam, 2025-05-15:
On web you can have different tabs or browser profiles loaded side by side. We may add an option in the future to combine two systems into one (add up energy values).

closed 2025-05-25 by zigam