netzero-labs/netzero #94: Powerwall ignoring Netzero

NDGLB03, 2025-05-16:
Well, this was odd.

Last night I had two half hour Octopus free energy slots so I set up automations to charge and stop charging at the start/end of each.

The first to set it to charge worked fine (20:30hrs).
At 21:00 hrs I receceived the e-mail saying the automation to set the reserve to 20% had worked - but the Powerwall ket charging.
I set up an additional automation for 21:05 hrs to set it to 20% but it kept charging. The battery was around 60%.

As my house load wasn't that great I went into the Tesla app and went off grid until the second free energy slot at 22:00hrs. I had to go to be so I don't know exactly what happened but it looks like the battery went back to 20% backup around 22:40hrs, not 22:30 as reported by the e-mail.

My normal reset at 05:30hrs to 20% this morning worked fine.

Was there some sort of outage last night?

Is there another way to stop the Powerwall charging other than going off grid?

zigam, 2025-05-17:
No issues on our end. In some cases, it can happen that Tesla fails to pass the configuration change to the Powerwall. There is unfortunately no way for Netzero to verify that. If there are network connectivity issues or the gateway is in a bad state, that's more likely to happen. If there issue persists, I would restart the gateway and maybe your wireless access point.

To stop charging your Powerwall, you can also manually reduce the backup reserve below its state of charge, using the Tesla app.

NDGLB03, 2025-05-17:
It's all working fine now so I guess there was a problem in the cloud somewhere.

I barely use the Tesla app nowadays and couldn't see how to reset the backup reserve but I've found it now.

Many thanks.