netzero-labs/netzero #95: grid_charging from API not working ?

rob0101, 2025-05-21:
curl -s -H "Authorization: Bearer $API_TOKEN" -H "Content-Type: application/json" --data '{"grid_charging":true}'  https://api.netzeroapp.io/api/v1/$SITE_ID/config

{"backup_reserve_percent":34,
"operational_mode":"self_consumption",
"energy_exports":"battery_ok",
"grid_charging":false,

: : : 

Is that meant to work ?

zigam, 2025-05-21:
It's meant to work, but for some reason it's not available for all systems. Do you see a Grid Charging option in your Tesla app (Settings > Powerwall)? Documentation: [Advanced Settings ](https://www.tesla.com/support/energy/powerwall/mobile-app/advanced-settings).

The option (when available) can be used to turn off grid charging in Time-Based Control mode.

rob0101, 2025-05-21:
Thanks for the reply.

No I don't have that option in my powerwall settings.  :-(

zigam, 2025-05-21:
I believe the availability of grid charging is configured in Tesla One, so you could check with your installer about changing that.

rob0101, 2025-05-21:
I've always been able to grid import and need to continue to do so.   I'd only want to disable it at specific times.

I'm told it's possible with the fleet api, but I don't have direct access to try that method.

zigam, 2025-05-21:
> I've always been able to grid import and need to continue to do so. I'd only want to disable it at specific times.

Right, that's what the Tesla One setting enables, see the Site Import Permissions setting here:

https://energylibrary.tesla.com/docs/Public/EnergyStorage/Powerwall/General/Compliance/PowerControlSystemFeatures/en-us/GUID-6ABE100F-7C90-408E-ABD4-D7249BBF32F3.html

It's a bit confusing though, because there's clearly also an undocumented "grid charging allowed but not configurable" state.

> I'm told it's possible with the fleet api, but I don't have direct access to try that method.

`disallow_charge_from_grid_with_solar_installed` is what Netzero uses under the hood when you use the API. I'm just not sure if it has an effect without the Tesla One setting above, but you can give it a try (meaning, you can test if the `grid_charging: false` API call impacts the behavior of your Powerwall).