netzero-labs/netzero #100: IOG - Toggle does not appear if Octopus Account Id is not uppercase

nickcopley-sa, 2025-05-27:
GIVEN a valid Octopus account number of A-AAAA1111
AND the API key is valid
WHEN the application loads the account details
THEN the "Automate Octopus Intelligent" toggle appears 👍 

GIVEN a valid Octopus account number of A-AAAa1111
AND the API key is valid
WHEN the application loads the account details
THEN the "Automate Octopus Intelligent" toggle appears 👎

zigam, 2025-05-27:
That would work if all account numbers are upper case. I can't find documentation that specifies that, so the safest thing to do is to send the account number as provided by users, and for users to copy the account number from Octopus verbatim.

nickcopley-sa, 2025-05-27:
I can't either. However the tariff lookup is not case sensitive it finds that information ok. It's in the code that must check if it's IOG. So you could use the account number format returned from the first call

zigam, 2025-05-27:
I'm not sure if that's what you're referring to, but the Octopus tariff list in Netzero is not using authenticated API calls -- account number and API key are not used for those API calls.

closed 2025-06-11 by zigam