netzero-labs/netzero #109: NetZero app unable to start EV charging with button

Danielhh47, 2025-06-10:
I was able to get solar charging working, but if I try to manually initiate a charge using the "Start Charging" button, nothing happens.

Danielhh47, 2025-06-11:
> Thanks for the report. Is this with an OCPP EV charger? Which brand? And
> how was charging stopped?
>
> With OCPP, charging starts automatically when plugging in. If it's stopped
> by the vehicle, the charger will not override that and fail to start again.

I resolved my issue. It was because the minimum OCPP amperage is 6 amps
(1.4kw @240vac) but the vehicle I was trying to charge accepts a minimum of
8 amps.

The vehicle would attempt to start charging, but stop almost immediately.
Then refuse to start charging again. The issue was entirely my vehicle, not
NetZero itself.

Having the ability to modify the lowest charging amperage would help with
this, but I imagine it's such a rare issue given most vehicles support
6amps as the lowest charging amplitude.

For reference, the vehicle I was using is a 2015 fiat 500e. The
manufacturer stated minimum charging amplitude at 240v is 12 amps, though I
have had success at 8 amps in my tests. At 6 or 7 amps it faults out with a
"service charging system" fault.

zigam, 2025-06-11:
Thanks for the follow-up! I suspect the issue might be common, so we will consider adding the minimum charge rate option.

closed 2025-07-08 by zigam