netzero-labs/netzero #111: multiple Tesla vehicles

simonog, 2025-06-11:
I have a Tesla Model S, a Model X and a Roadster.  Each has its own Tesla wall charger.  Can NetZero handle charging all of them either together or in sequence when there is excess solar (or cheap power)?

zigam, 2025-06-12:
Are these pre-2021 Model S/X, which are incompatible with Charge on Solar?

If you pair the vehicles in Netzero, you can automate starting and stopping charging, but it won't handle multiple vehicles very well (only the first vehicle will start charging).

simonog, 2025-06-16:
Magic if it can though I suspect Tesla doesn’t offer the feature

zigam, 2025-07-24:
Tesla recently added Fleet Telemetry support for Model S/X vehicles with Intel Atom computers. This is vehicles with MCU2 (2018+) or older vehicles with an Infotainment Upgrade (MCU1->MCU2). This requires firmware version 2025.20 or later.

I'm wondering if this change also enabled Charge on Solar for those vehicles?