netzero-labs/netzero #113: Octopus Tariff Enhancement Request

ToastyOasty, 2025-06-13:
I'm a very happy user of NetZero - so happy that I've already subscribed to the paid service.  

As an Octopus Intelligent Go customer, the NetZero automation is excellent and very reliable.  Is it possible though to consider an further option where, when enabling the automation, there could be an option to prevent battery drain, as exists in the manually setup automations?

I say this becuase, often when plugging in my car to charge, the half hour Octopus charging slots kick in, and NetZero (as it's supposed to) then tries to fully charge my Powerwalls, at the same time.  I often don't want to do this, as I know I'll likely have solar to complete that charge - and as a result I find myself exporting that solar to the grid, for which I don't get paid.  

As a result, I find myself overriding NetZero to prevent this charge

If there's another way to configure this, using the Automation, sorry, I'm missing this.

zigam, 2025-06-13:
Thanks for the suggestion and for subscribing!

Configurable IOG integration that will preserve Powerwall charge (instead of charging it up) is on the roadmap!

JoeDMB, 2025-07-24:
Looks to have now been released so have switched it on.  When the sun eventually appears again, will be good to test

zigam, 2025-07-24:
> I noticed this, installed the update and tested today - all appears to be working perfectly!

Great, thanks for the update!