netzero-labs/netzero #114: Automation time always reset to 12:00

AurelioC52, 2025-06-13:
Trying to make a schedule automation triggering at a certain time of day (changing the hour & minutes in the upper left box) before selecting one action, the automation created always revert to 12:00 AM.

What I'm doing wrong ?

Thank you.

zigam, 2025-06-13:
Thanks for the report. Is this on iOS, Android, or web? Which app version (Settings / About Netzero)?

zigam, 2025-06-13:
Still can't reproduce the 12:00 AM issue, which browser are you using and what are you entering in the text box?

If the Tesla app is not showing data, there's likely an issue with network connectivity from your gateway or Powerwall 3. Netzero will show the same data as Tesla.

zigam, 2025-06-14:
I see, the issue is with the Mac app. I can reproduce the issue, we will fix that. As a workaround, you can press **Return** after entering the time.

AurelioC52, 2025-06-14:
> zigam

Thanks. 
I’ll try that but before I need to fix the communications behaviour of the gateway that suddenly started yesterday, and I’m afraid I will need the installer to come.

Again, thank you very much.

zigam, 2025-07-08:
Fixed in version 2.1.2.