netzero-labs/netzero #117: OCPP Connection verification error with Autel charger app

C0deMonki, 2025-06-22:
I am getting the following connection verification error with the Autel Charge app (iOS).  I am copying and pasting the OCPP provider URL (Server URL in the Autel app) and Charge Point ID (Charger ID in Autel app) from NetZero.  Leaving the Authorization Key and Certificate blank.

[image]
[image]

Any help would be greatly appreciated.

zigam, 2025-06-22:
From our experience, Autel doesn't support secure connections. Try replacing `wss://` with `ws://`

zigam, 2025-06-23:
The certificate is valid and shouldn’t need to be added, I suspect secure networking support is just missing on Autel’s side.

closed 2025-07-08 by zigam