netzero-labs/netzero #119: Intelligent Octopus Go Charges Full Price in Scheduled Slot

rfctabs, 2025-07-01:
Hi, 

In NetZero, I switched on the Automate Intelligent Octopus Go feature. When I plugged in my EV, and set completion time for the next morning. Octopus created a charging schedule with time slots running, on and off, until 04:00. When a time slot arrived, NetZero set the Powerwall 3 to a backup reserve of 100% as expected, and back to 20% at the end of the slot. This happened even when the EV was not charging, so the Powerwall was still being charged even though the EV was not.

I just reviewed the charges to my account from Octopus, and I have been charged the full day rate to charge the Powerwall if the EV was not actively charging. Is this how it works? Or have I missed a setting somewhere? Or do I need to challenge Octopus as to why they billed me this way?

Thanks,

Peter

rfctabs, 2025-07-02:
Any update on this?

zigam, 2025-07-03:
Thanks for the report. Based on what we've heard from IOG users, Octopus can take a few days to update the rates in their app. If the rates don't get updated after a few days, please contact Octopus.

closed 2025-07-08 by zigam