netzero-labs/netzero #120: Authentication Issues

lb74-ghub, 2025-07-02:
Good Morning from the UK. As of last night, I am unable to use the Net Zero app. Initially, the app states Offline Mode Diagnostics Only and then eventually times out. Re-authenticating with my Tesla login details produces an authentication error claiming 'Invalid token' and a 500 server error code. Any ideas? Thanks.

jwt99412, 2025-07-02:
Me too. Was alerted by x post then tried the app and get the same behaviour

zigam, 2025-07-02:
There is an issue with authentication on Tesla's side, which will affect Netzero app and automations. We're working with Tesla on getting this resolved.

jwt99412, 2025-07-02:
Thank you for the quick reply 😀

zigam, 2025-07-02:
Forgot to send an update here, but issue was resolved starting July 2 12pm UTC.

zigam, 2025-07-03:
> Is there a way to get the automations to retry if they failed?

Automations will automatically retry, many times. However with a prolonged outage like this, the retries will eventually give up since it's no longer safe to assume the automation can still be run (e.g. too much time has passed for a scheduled automation, or too much time passed since an event triggered).

IanCassTwo, 2025-07-03:
Understandable, but in this case it resulted in my Powerwall remaining at 100% reserve and my house using expensive peak-time grid import. Currently it's set up as a time-based automation "at 4:55am set reserve to 10%". Is there a way to set it as "during hours of 4:55am and 11:59pm ensure reserve is set to 10%" where the condition is retested and reapplied if necessary on a regular basis during this period?

zigam, 2025-07-04:
> Understandable, but in this case it resulted in my Powerwall remaining at 100% reserve and my house using expensive peak-time grid import. Currently it's set up as a time-based automation "at 4:55am [...]

This is not possible with in-app automations, however it could be accomplished using the [Netzero API](https://docs.netzero.energy/docs/tesla/API).

Note that this recent outage is the first such major outage in the last 2 years.

zigam, 2025-07-04:
> Not connected with the recent outage / auth issue, I've had similar issues where the automation hasn't completely set the parameters as planned. sometimes I think my Powerwall 2 gets confused and chooses to ignore changes from TBC to SP and vice versa, or with the % shift (I'm either SP 0% or TBC 100%, rarely anything different.) So maybe some sort of ability as suggested to allow the automation to recheck / reapply a couple of times every minute or so......and maybe an option for us to tell it 'not to' if we are choosing to manually interfere with the settings. Just my twopence.

Netzero does retry, recheck, and in fact apply every automation twice. Unfortunately Tesla's side is not very reliable, and sometimes still fails to apply automations (in ways that can't be verified by Netzero).

closed 2025-07-08 by zigam