netzero-labs/netzero #122: Automation not running.

Serenita247, 2025-07-08:
I am trying to set up automation on my powerwalls  through the winter period so that by 4 pm daily I have a minimum charge in them to cover me in my peak time between 4 and 8 pm. My aim is to grid charge over the solar charge to get them at sat 45% for example. My routines won’t run yet I know the powerwalls can grid charge by turning on storm watch. I also can’t log into the gateway under the monitoring tab using the correct password. I have held in the reset button on the gateway and it reboots. At a loss of what I am doing wrong or there is an issue. It also makes no difference if I set the Tesla app to self powered or TOU. Hopefully someone can find an error in my automation or if there is an issue with the gateway. I am an end user and installer so I have access to Powerhub also. 

[image]

zigam, 2025-07-08:
To charge your Powerwall to at least 45%, just set the backup reserve to 45%, no other actions required. Then a few hours later, reset the backup reserve to your normal value. So for example:

Every day at 2:00 PM: Set backup reserve to: 45%.
Every day at 4:00 AM: Set backup reserve to: 20%.

More details here: https://docs.netzero.energy/docs/tesla/Automation

For troubleshooting diagnostics, see the documentation for Powerwall 2 and 3:
https://docs.netzero.energy/docs/diagnostics/Powerwall2
https://docs.netzero.energy/docs/diagnostics/Powerwall3

Serenita247, 2025-07-10:
Zigam

Thanks for that I followed your directions and have tested for a few days and it works like a treat. Now that I have it up and running I have another question. My home is totally automated with my HWS running of a Wi-Fi switch that control with Alexa and Smart life. Is there a way I can either :

*	Integrate Net Zero with one of the apps to turn on the HWS when the battery is full or
*	Wire a low voltage relay off the AUX contacts in the Tesla Gateway with this existing relay and have Net Zero switch the AUX dry contact on when the battery is full and exporting so I can divert export power to heat the HWS or
*	Is there a brand of relay out there that integrates directly with Net Zero

 

 

Regards,

 

David Borg.

 

Borg Power Solutions PTY Ltd T/A 
138B Dowling St

Falls Creek NSW 2540 
Mobile: 0412939709      

www.borgpowersolutions.com.au 
Licenced Electrical Contractors.

Solar Installations.

Service Provider.                                                                                                   Leave us a review.

             <https://www.facebook.com/David-Borg-Electrical-Solar-105553254562621> 

 

 

The information contained in this email message is privileged and confidential, and is intended only for use of the addressee.

If you are not the intended recipient, you are hereby notified that any disclosure, reproduction, distribution or other use of this

Communication is strictly prohibited. If you have received this communication in error, please notify the sender by reply

Transmission and delete the message without copying or disclosing it.

 

https://docs.netzero.energy/docs/diagnostics/Powerwall2
https://docs.netzero.energy/docs/diagnostics/Powerwall3

zigam, 2025-07-10:
We don't have any home automation integration currently, but that's something we're exploring for future. In the meantime the [Netzero API](https://docs.netzero.energy/docs/tesla/API) can be used for automation.

Serenita247, 2025-07-10:
Is this the current automated setups or is this something different?

 

 

  _____

zigam, 2025-07-10:
Netzero API is different, have a look at the linked documentation.

Serenita247, 2025-07-10:
Is this the current automated setups or is this something different?Get Outlook for iOS From: Ziga Mahkovec ***@***.***>Sent: Thursday, July 10, 2025 8:03 pmTo: netzero-labs/netzero ***@***.***>Cc: Serenita247 ***@***.***>; Author ***@***.***>Subject: Re: [netzero-labs/netzero] Automation not running. (Issue #122) zigam left a comment (netzero-labs/netzero#122)We don't have any home automation integration currently, but that's something we're exploring for future. In the meantime the Netzero API can be used for automation.

closed 2025-07-13 by zigam