netzero-labs/netzero #126: setting up an automation

JonMurphy, 2025-07-12:
Continued from [here](https://github.com/netzero-labs/netzero/issues/121)

I am still experiencing problems.  I of the graphing seems to work as expected.  But not the automations.  Here are my two attempts.

The 1st one:
Clicked on Edit Automation Time (I didn't know any better)

Every Tue at 2:00 AM:
Set backup reserve to: 80%;
Set operational mode to: Self-Powered;
Set energy exports to: Everything (solar and battery);
Set grid charging to: Enabled.

The 2nd one:
When electricity price is below $0.09 (between 2:00 AM and 2:30 AM):
Set backup reserve to: 80%;
Set operational mode to: Time-Based Control;
Set grid charging to: Enabled.

The price this morning was < $0.09

[image]

zigam, 2025-07-13:
Netzero automations use so called threshold-crossing triggers. They will run at the moment when the threshold is crossed, in this case at the moment the price drops from at or above 9c to below 9c. If you additionally restrict that to a time frame (2-2.30am) it's easy to miss the moment when the threshold is crossed, and the automation won't run.

Can you describe your goal with this automation?

JonMurphy, 2025-07-13:
It is pretty much as written:
• When electricity price is below $0.09 (between 2:00 AM and 2:30 AM):
• Set backup reserve to: 80%;
• Set operational mode to: Time-Based Control;
• Set grid charging to: Enabled.

I set to from 2:00 AM and 2:30 AM **to test it**.  Ultimately it will be 2 AM to 4 AM.
I set the electricity price to $0.09 **to test**.  This will end up at $0.05.

And I'd like to add one more line: if the SoC is < 50% (if this is possible).

zigam, 2025-07-14:
If all the prices are below 9c, the automation will not trigger. The automation trigger when the price **falls below** 9c. And then you would generally also run a related automation (usually to reset the state) when the price rises back above 9c.

JonMurphy, 2025-07-14:
OK.  Now I understand.  Thank you!  Thank you!

Feature Request:  Make a non-threshold based version for current price.  If current price is < 9¢ then allow charging.

closed 2025-09-02 by zigam