netzero-labs/netzero #127: Netzero not changing the charging speed

jultican024, 2025-07-12:
I just setup NetZero with a ELECQ Home EV Charger via OCPP.  It shows up in NetZero as a compatible EV charger.  I have charge on solar enabled and to wait for my Powerwall 2 to fully charge first.

When a charge session is started the wall charger doesn't fluctuate the charging speed to only use the excess solar?  I couldn't find any logs or any other documentation to help with the troubleshooting.

zigam, 2025-07-13:
Thanks for the report, we haven't had any ELECQ chargers connect so far, so it's possible there are incompatibilities. We'll look at the logs on our end and report back.

jultican024, 2025-07-27:
Any update on this issue?  I have added a second charger to might system.

zigam, 2025-07-29:
Sorry for the delay. We looked at the data and the issue is that the EV charger doesn't report the metering values that Netzero requests (e.g. `Current.Import` or `Current.Offered`). These metering values are required to modulate charging speed when charging on solar.

Looks like ELECQ's OCPP implementation is incomplete.

zigam, 2025-07-31:
> If you want to include "Current.Import" in the reported Measurands, the CSMS needs to send a ChangeConfiguration message with {key = MeterValuesSampledData, value including Current.Import}.

Netzero does configure that key, here is the current configuration for the charge point in question:

```
'MeterValuesSampledData': 'Current.Import,Current.Offered,Energy.Active.Import.Register,Energy.Active.Import.Interval,Frequency,Power.Active.Import,Power.Offered,SoC,Temperature,Voltage'
```

But the configuration seems to be ignored, the only values returned are `Energy.Active.Import.Register` and `Power.Active.Import`.

zileyue163, 2025-08-01:
I'll check the Charger's log to see what the issue is.

zileyue163, 2025-08-01:
> Netzero does configure that key, here is the current configuration for the charge point in question:
>
> ```
> 'MeterValuesSampledData': [...]

We checked the logs, and the CSMS did send a ChangeConfiguration message. However, some Measurands in the Value field are not in our supported list, so we responded with Rejected. The reason was: "Given value is not valid".

Received <<<:
[2,"849d80f6-fdbc-4bce-bdb0-6e0a23dedca1","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import,Current.Offered,Energy.Active.Import.Register,Energy.Active.Import.Interval,Frequency,Power.Active.Import,Power.Offered,SoC,Temperature,Voltage"}]

Sent >>>:
[3,"849d80f6-fdbc-4bce-bdb0-6e0a23dedca1",{"status":"Rejected"}]

Could you send only the Measurands that are in our supported list? For example, like this:

[2,"849d80f6-fdbc-4bce-bdb0-6e0a23dedca1","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import, Energy.Active.Import.Register, Frequency, Power.Active.Import, Voltage, SoC"}]

jultican024, 2025-08-11:
> We'll give this a try. We would also suggest `Current.Offered` be added to your supported list, since it's an allowable measurand according to the OCPP 1.6 spec. There is no way for Netzero to confirm which measurands a charge point supports.
> 
> (Even if the values are not actually reported -- there are other supported measurands like this, e.g. `SoC`.)

Any update on this?  Do I need to do anything to enable this new functionality?

zigam, 2025-08-11:
> Any update on this? Do I need to do anything to enable this new functionality?

We made the change, so if you restart your EV charger, we can confirm that the meter values are correct.

We still need to implement a workaround for the missing `Current.Offered` values.

zigam, 2025-08-13:
The meter values at least are getting sent now. As I mentioned we would still need a workaround for the missing `Current.Offered` values.

But there appears to be another (larger) issue. `SetChargingProfile` commands to set the charging rate (which is also how charging is stopped) are being rejected. Perhaps @zileyue163 can shed some light on this.

zileyue163, 2025-08-14:
> The meter values at least are getting sent now. As I mentioned we would still need a workaround for the missing `Current.Offered` values.
>
> But there appears to be another (larger) issue. [...]

I guess you may have confused the two chargers. You have two chargers on hand: AU105B1G000219 and AU105B1G000028.

During your testing, AU105B1G000028 was in charging, while AU105B1G000219 was not. However, you sent the following messages to AU105B1G000219:

[2,"76f26160-df09-40bd-823d-5fd4cebd163a","RemoteStopTransaction",{"transactionId":230255}]

[2,"dac56e7f-092b-4803-a22e-2a67ac5dd232","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":230255,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]

Since AU105B1G000219 was not in charging, both the RemoteStopTransaction and SetChargingProfile messages were rejected.

jultican024, 2025-08-14:
> [@jultican024](https://github.com/jultican024) do you have both chargers configured with the same Charge Point ID? This is not supported currently in Netzero, so you would have to turn off OCPP on one of them while testing this.

I will get one of them reset.  Does Netzero support multiple OCPP chargers?

jultican024, 2025-08-14:
@zigam I have factory reset the secondary unit (AU105B1G000219) and confirmed it is no longer pointed at the NetZero OCPP server.  I then rebooted the first charger (AU105B1G000028) and it seems to be registered with the site. 

I clicked the start charging button:

August 14, 2025 4:13:14 PM
Set backup reserve to the current state of charge (preserve Powerwall charge)
Vehicle charging started

It seemed to turn back off 16 seconds later for some reason.  (The vehicle was not fully charged)
August 14, 2025 4:13:30 PM
No action
Vehicle charging stopped

Times are in PDT.

zigam, 2025-08-16:
> It seemed to turn back off 16 seconds later for some reason. (The vehicle was not fully charged)

The last missing part is a workaround for the missing `Current.Offered` value, I'll send an update once that's added.

zigam, 2025-08-19:
@jultican024 We added a workaround for `Current.Offered`, please give this another try. Let us know the time & date when you start the Charge on Solar session, so we can look at the logs.

jultican024, 2025-08-19:
It seemed to have reset my account?  I lost all my configurations and automation along with the charge point ID being different.  I factory reset the EV charger and re-setup the OCPP info.  (SDDE-3529)

jultican024, 2025-08-19:
Charge on Solar kicked in about 0916 on 08/19/2025.  Kind of a cloudy day here so the session started and stopped a few times.  Seems to been stable since 0925.  It is looking like your work around is successful at the moment.

jultican024, 2025-08-20:
> > Other OCPP chargers have configuration that allows them to skip "authorization" and start charging as soon as the vehicle is plugged in. There's no OCPP specification for this, but different chargers will use: `AuthEnabled=false` (Wallbox), `FreeCharging=True` (Zappi), or `FreeVend=true` (Autel).
> > [@zileyue163](https://github.com/zileyue163) is there a similar configuration option for ELECQ?
> 
> Yes, for ELECQ charger, the "Autostart" function can be enabled in the ELECQ app to allow the charger to start charging immediately when the connector is plugged in. If this function is not desired, the "Autostart" setting can be turned off. The "Autostart" setting is located within the "Access" button. ELECQ charger also has a "Scheduled Charging" function. This can be found under the "Schedule" button in the ELECQ app.This function can also be turned off.

I found the autostart feature and have enabled it. Thank you!

zileyue163, 2025-08-21:
> So, I found the autostart feature is not enabled or accessible in the app when the OCCP server is directed at a different server [@zileyue163](https://github.com/zileyue163). I can not get a session to start by simply plugging in the EV. I tried factory resetting the device and enabling it before moving the OCPP target but it gets disabled once that action is initiated.
> 
> [@zigam](https://github.com/zigam) I tried charging a non-telsa vehicle today and it looks like I spoke to soon. The charger goes to 100% and does not throttle as expected. :( Another issue I run into is when I click the Stop Charging button the Start Charging button does not re-appear? I have to reboot my EV charger to get the button back.
> 
> August 20, 2025 2:57:30 PM Set backup reserve to the current state of charge (preserve Powerwall charge) Vehicle charging started
> 
> <img alt="Image" width="395" height="145" src="https://private-user-images.githubusercontent.com/25798020/480246004-dc390c8e-237a-4768-8482-f947c8cd3348.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3NjEzNjMsIm5iZiI6MTc1NTc2MTA2MywicGF0aCI6Ii8yNTc5ODAyMC80ODAyNDYwMDQtZGMzOTBjOGUtMjM3YS00NzY4LTg0ODItZjk0N2M4Y2QzMzQ4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIxVDA3MjQyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUyMWYzZjI4MjE1MzE3NTRjYWJiZWNmZWQ0YTdjNWQ3YTM5ZDI2MTRhMTJjMmEyZTk1NmNiYWU1NWQxMWM0NmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gyHF__cf287vrDq5aCIxu8fn96iSW99yJ9jkGBYi4Dw">

When an ELECQ charger is switched to a third-party OCPP server, functions such as Autostart and Scheduled Charging will be disabled. This is because Autostart and Scheduled Charging rely on internally defined idTags that are not recognized by third-party OCPP servers. In other words, once an ELECQ charger is connected to a third-party OCPP server, features like Autostart and Scheduled Charging will no longer be available. Sorry for forgetting to inform you about this limitation in advance.

zileyue163, 2025-08-26:
> We will need to hear back from [@zileyue163](https://github.com/zileyue163) about the ability to set the charging rate.

Checking now...

zileyue163, 2025-08-26:
> I just kicked off a charge on solar session with my non-Tesla EV. The charger is charging at full capacity and not at the available excess solar. 😥
> 
> Event: Vehicle charging started. Time: Aug 25, 2025 at 1:38:32 PM.

Which charger are you using?  AU105B1G000219 or  AU105B1G000028 ? I'm a bit confused.

jultican024, 2025-08-26:
> > I just kicked off a charge on solar session with my non-Tesla EV. The charger is charging at full capacity and not at the available excess solar. 😥
> > Event: Vehicle charging started. Time: Aug 25, 2025 at 1:38:32 PM.
> 
> Which charger are you using? AU105B1G000219 or AU105B1G000028 ?

I am using AU105B1G000028 @zileyue163

zileyue163, 2025-08-26:
pls check AU105B1G000028 again. @jultican024

zileyue163, 2025-08-27:
> > pls check AU105B1G000028 again. [@jultican024](https://github.com/jultican024)
> 
> What am I checking for [@zileyue163](https://github.com/zileyue163) ?

Sorry for not making it clear earlier. Please test whether the charging current is controlled by the server's SetChargingProfile message (if your solar power supply capacity can be synchronized to the server, the server should send the corresponding SetChargingProfile message to the charger).

zileyue163, 2025-08-28:
> surpasses

I've checked the logs, and I can see that you were charging during the period from 2025-08-26 12:27:17 to 2025-08-26 16:05:32 in your local time. Throughout this charging session, the OCPP server sent multiple SetChargingProfile messages to the charger.
Each time, the charging station correctly limited the current according to the limit values specified in the SetChargingProfile messages, as shown in the logs below (since there were numerous SetChargingProfile messages, I've only extracted a few as examples).
From these logs, it is evident that the charger fully executed the SetChargingProfile commands. Therefore, if you mentioned that the charger's output exceeded the solar power supply capacity, we need to verify whether the solar power capacity has been properly synchronized to the server and whether the server's calculations are correct. This requires further investigation by @zigam.

1.1  SetChargingProfile limit 31 A:

[2025-08-26 12:30:18.405]: <<< [2,"e5cf2648-1d0f-4eb0-9e21-6c76dedc815f","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":251273,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":31}]}}}]

1.2 MeterValues L1 Current 31.117700 A:

[2025-08-26 12:30:47.701]: >>> [2,"7ca09099-23a2-45d4-9f62-5097ef26e92c","MeterValues",{"connectorId":1,"meterValue":[{"sampledValue":[{"context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A","value":"31.117700"},{"context":"Sample.Periodic","measurand":"Current.Import","phase":"L2","unit":"A","value":"0.000000"},{"context":"Sample.Periodic","measurand":"Current.Import","phase":"L3","unit":"A","value":"0.000000"},{"context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V","value":"237.021300"},{"context":"Sample.Periodic","measurand":"Voltage","phase":"L2","unit":"V","value":"0.000000"},{"context":"Sample.Periodic","measurand":"Voltage","phase":"L3","unit":"V","value":"0.000000"},{"context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"kWh","value":"327.802000"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","phase":"L1","unit":"W","value":"7369"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","phase":"L2","unit":"W","value":"0"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","phase":"L3","unit":"W","value":"0"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W","value":"7365"}],"timestamp":"2025-08-26T19:30:47Z"}],"transactionId":251273}]

2.1 SetChargingProfile limit 32 A:

[2025-08-26 13:01:13.226]: <<< [2,"c979f992-0ebb-4852-a8ff-bb0d80313abb","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":251273,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]

2.2  MeterValues L1 Current 32.007100 A:

[2025-08-26 13:01:23.185]: >>> [2,"e54c8c43-1dfd-4d37-a2d2-eb83bfabd72b","MeterValues",{"connectorId":1,"meterValue":[{"sampledValue":[{"context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A","value":"32.007100"},{"context":"Sample.Periodic","measurand":"Current.Import","phase":"L2","unit":"A","value":"0.000000"},{"context":"Sample.Periodic","measurand":"Current.Import","phase":"L3","unit":"A","value":"0.000000"},{"context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V","value":"234.384000"},{"context":"Sample.Periodic","measurand":"Voltage","phase":"L2","unit":"V","value":"0.000000"},{"context":"Sample.Periodic","measurand":"Voltage","phase":"L3","unit":"V","value":"0.000000"},{"context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"kWh","value":"331.652000"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","phase":"L1","unit":"W","value":"7485"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","phase":"L2","unit":"W","value":"0"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","phase":"L3","unit":"W","value":"0"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W","value":"7488"}],"timestamp":"2025-08-26T20:01:22Z"}],"transactionId":251273}]

3.1 SetChargingProfile limit 0 A:

[2025-08-26 14:00:33.073]: <<< [2,"510e723e-2e94-4153-a089-a9b82a1c5a88","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":251273,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]

3.2 MeterValues L1 Current 0 A:

[2025-08-26 14:01:02.533]: >>> [2,"a67e0b5b-a0d5-4944-9391-cebb9b493cda","MeterValues",{"connectorId":1,"meterValue":[{"sampledValue":[{"context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A","value":"0.000000"},{"context":"Sample.Periodic","measurand":"Current.Import","phase":"L2","unit":"A","value":"0.000000"},{"context":"Sample.Periodic","measurand":"Current.Import","phase":"L3","unit":"A","value":"0.000000"},{"context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V","value":"237.577800"},{"context":"Sample.Periodic","measurand":"Voltage","phase":"L2","unit":"V","value":"0.000000"},{"context":"Sample.Periodic","measurand" [... cut ...] Register","unit":"kWh","value":"340.539000"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","phase":"L1","unit":"W","value":"2618"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","phase":"L2","unit":"W","value":"0"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","phase":"L3","unit":"W","value":"0"},{"context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W","value":"2617"}],"timestamp":"2025-08-26T22:02:12Z"}],"transactionId":251273}]