netzero-labs/netzero #138: Automation for Rivian charging on Tesla Wall Conmector

samsonblackwell, 2025-08-05:
Hi there. I tried to find documentation on this: can I set up my Gen 3 Universal Wall Connector to auto magically start/stop charging my Rivian using the Netzero app?

zigam, 2025-08-05:
Yes, you can issue Start/Stop commands with the wall connector. You can test this first in Settings > Vehicle Charging, and then add automations with these actions.

samsonblackwell, 2025-08-05:
When I press “Start Charging” in the Settings > Vehicle Charging page, I get a spinning icon for a moment, but the vehicle does not start charging. Sent from my iPhone

zigam, 2025-08-05:
A few things to confirm:

- Does this work if you start charging from the Tesla app?
- It may depend on how charging was stopped. If it's the Rivian that stopped charging, the Wall Connector might not be able to resume.
- Also, it might be necessary to wake up the Rivian first.

samsonblackwell, 2025-08-09:
Confirming that I can start from the Tesla app if I last stopped from the Tesla app. If I stop with the Rivian it won’t allow start from the Tesla app. Sent from my iPhone

closed 2025-08-12 by zigam