netzero-labs/netzero #140: Get/Update Automations via API

Carl-Bradbury, 2025-08-08:
I'm loving NetZero right now, and I have connected to the API to read and update the current config, but is there any way to use the API to amend the configuration of my automations?

Carl-Bradbury, 2025-08-10:
Thanks for the swift reply!
I noticed that the curve of prices available through the Octopus Agile tariff changes so much that the optimum rates at which to start/stop charging/discharging are different each day. I have written a script to work out these rates but I need to manually edit the NetZero automations with the new rates. I was hoping to use the NetZero API to update the automations and then I can simply run the script once per day to keep my system optimised.

carlbradbury, 2025-08-11:
If I did this, could I control the settings through the Tesla Fleet API directly?

PG-code404, 2025-11-23:
Tesla App only has fixed tariff set up using Utility rate plan and can't look up Octopus Agile so dynamic tariffs needed a third party automation.

zigam, 2025-11-24:
> Tesla App only has fixed tariff set up using Utility rate plan and can't look up Octopus Agile so dynamic tariffs needed a third party automation.

Netzero can update Agile tariffs automatically, without the need for automations or API support.